aldas/modbus-tcp-client #1: I'm getting a wrong packet on fc03

Brugui7, 2017-08-08:
With this code:

[image]

I'm getting this error

```An exception occurred 
Message -> Read total timeout expired 
#0 C:\xampp\htdocs\modbus-tcp-client\src\Network\BinaryStreamConnection.php(137): ModbusTcpClient\Network\BinaryStreamConnection->receive() #1 C:\xampp\htdocs\modbus-tcp-client\functions\Read.php(25): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive(Object(ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest)) #2 {main}```

Could you help me?

Btw, this repo is about directly modbus TCP or above RTU modbus TCP ??

Nice repo!!!

aldas, 2017-08-08:
Hi, this is modbus over tcp (without crc part) http://www.simplymodbus.ca/TCP.htm 

From error I can only see that this lib was able to connect to modbus master but did not get respond and therefore request timeouts. It could be as you suspect - you may have different protocol for your plc. Atm only thing that I can recommend is to try same fc3 query with same ip/tcp/port with https://github.com/MightyPork/phpmodbus library.

aldas, 2017-08-08:
just to be sure you could start debuggin by checking if you actually can connect from your pc to plc modbus port

Under windows:
```
telnet <modbus ip> <port>
```
it should not give errors.

aldas, 2017-08-08:
and have you set php error level to be strict enough? from that picture it seems to me that you are using phpstorm and 'ReadHoldingRegistersRequest' class has not been correctly imported (gray background). It is probably not the reason for you error as for me it results ```Fatal error: Uncaught Error: Class 'ReadHoldingRegistersRequest' not found in ```

Brugui7, 2017-08-08:
Thank you very much for your help and fast response!

I'll try what you've say.

closed 2017-08-08 by Brugui7