aldas/modbus-tcp-client #9: Question: The project support ModBus RTU?

rfsales, 2018-07-10:
The project support ModBus RTU? 
If not there is any way you guys have already tryed that situation with php?

My needs: A TCP connection using Modbus RTU

aldas, 2018-07-11:
Hi, this project does not support out of the box Modbus RTU but this does not mean that you could not make it work.

As I understand difference between RTU and TCP/IP is that:
1. RTU header contains only slave id. TCP/IP header contains of transaction id, protocol id, length, unitid
1. RTU packed has CRC appended

see this (http://www.simplymodbus.ca/TCP.htm)
[image]

so you could of use library to:
1. create TCP/IP package.
2. remove tcp header
3. add RTU slaveid at the start of packet
4. calculate CRC and add at the end of packet
5. send request

and when you receive response
6. remove slave if from the beginning (1 byte)
7. remove CRC from the end (last 2 bytes)
8. add made up TCP/IP header and parse response with this library

aldas, 2018-07-11:
Maybe this works

```
<?php

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ResponseFactory;
use ModbusTcpClient\Utils\Types;

require __DIR__ . '/../vendor/autoload.php';

// stolen from https://github.com/toggio/PhpSerialModbus/blob/master/PhpSerialModbus.php#L65
function crc16($data)
{
    $crc = 0xFFFF;
    for ($i = 0, $iMax = strlen($data); $i < $iMax; $i++) {
        $crc ^= ord($data[$i]);
        for ($j = 8; $j != 0; $j--) {
            if (($crc & 0x0001) !== 0) {
                $crc >>= 1;
                $crc ^= 0xA001;
            } else
                $crc >>= 1;
        }
    }
    $highCrc = floor($crc / 256);
    $lowCrc = ($crc - $highCrc * 256);
    return chr($lowCrc) . chr($highCrc);
}

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->build();

$startAddress = 256;
$quantity = 6;
$slaveId = 1;

$packet = substr((string)new ReadHoldingRegistersRequest($startAddress, $quantity), 6);
$packetWithCrc = Types::toByte($slaveId) . $packet . crc16($packet);

try {
    $binaryData = $connection->connect()->sendAndReceive($packetWithCrc);
    echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

    $binaryData = substr($binaryData, 1, -2); // remove slaveId and crc
    $tcpHeader = "\x00\x00\x00\x00\x00" . Types::toByte(strlen($binaryData)) . '\x00'; // create dummy packet header
    $response = $tcpHeader . $response;

    $response = ResponseFactory::parseResponseOrThrow($binaryData);
    echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
    echo 'Data parsed from packet (bytes):' . PHP_EOL;
    print_r($response->getData());

} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}
```

rfsales, 2018-07-12:
I have no words!!
Will try. 

Thank you very much!

aldas, 2018-07-13:
@rfsales let me know if you got it working. If it works I could add it to this library. 

If it does not work maybe you could sniff network traffic (some rtu packet that is sent some other Modbus RTU client and its response) and I might help you make it work with this lib.

rfsales, 2018-07-14:
@aldas Of course I will let you know!

I´m setting up my environment, it will be something like:
[image]

aldas, 2018-07-14:
Try this branch #10

rfsales, 2018-07-16:
First Try:

An exception occurred Read total timeout expired 
#0 /var/www/html/modbus-tcp-client-master/src/Network/BinaryStreamConnection.php(91): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom(Array, 0.3, NULL) 
#1 /var/www/html/modbus-tcp-client-master/src/Network/BinaryStreamConnection.php(112): ModbusTcpClient\Network\BinaryStreamConnection->receive() 
#2 /var/www/html/modbus-tcp-client-master/examples/rtu.php(21): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive('\x01\x03\x00\x05\x00\x06\xD5\xC9') 
#3 {main}

aldas, 2018-07-16:
allright,

first try increasing read timeout
```
$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->setReadTimeoutSec(5) // increase read timeout to 5 seconds
    ->build();
```

then try timeout between sending data and receiving

replace `$binaryData = $connection->connect()->sendAndReceive($rtuPacket);`  with

```
    $connection->connect()->send($rtuPacket);
    usleep(300000); // sleep 0.3 seconds
    $binaryData = $connection->receive();
```

rfsales, 2018-07-16:
Binary received (in hex): 0304fa0f00067131 
Parsed packet (in hex): 686b000000fd0304fa0f0006 
Data parsed from packet (bytes): Array ( [0] => 15 [1] => 0 [2] => 6 )

aldas, 2018-07-16:
could you paste code for the lower 'RTU received part.  That binary does not make sense atm. Is it FC3 or FC4? 

it does not seem to be valid packet even for RTU. From http://www.simplymodbus.ca/FC04.htm response should be

1 byte for slave id 
1 byte for function code
1 byte for number of data bytes to follow
N data bytes
2 byte for CRC16

but for `0304fa0f00067131` this does not add up

03 - is slave id
04 - function code
FA - data bytes (FA is 250 in dec)
... 250 bytes of data
7131 - crc16

rfsales, 2018-07-16:
I'm troubleshooting locally, maybe the cable or connector.
Will let you know.

rfsales, 2018-07-16:
SUCESSS!

Wrong configuration: It is RS232 and it was RS485 and changed the broken cable.

Binary received (in hex): 01030c002300330052006f00000000d0c9 
Parsed packet (in hex): c5f30000000f01030c002300330052006f00000000 
Data parsed from packet (bytes): Array ( [0] => 0 [1] => 35 [2] => 0 [3] => 51 [4] => 0 [5] => 82 [6] => 0 [7] => 111 [8] => 0 [9] => 0 [10] => 0 [11] => 0 )

rfsales, 2018-07-17:
Current code:

```
<?php

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;
use ModbusTcpClient\Packet\RtuConverter;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(40003) //GPRS Server Port
    ->setHost('192.168.2.235') //GPRS Server Address
	->setReadTimeoutSec(5) // increase read timeout to 5 seconds
    ->build();

$startAddress = 5; //5
$quantity = 6; //6
$slaveId = 1; // RTU packet slave id equivalent is Modbus TCP unitId

$rtuPacket = RtuConverter::toRtu(new ReadInputRegistersRequest($startAddress, $quantity, $slaveId));

try {
	$connection->connect()->send($rtuPacket);
    usleep(3*1000000); // sleep 0.3 seconds
    $binaryData = $connection->receive();
	
    echo '<br>Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

    $response = RtuConverter::fromRtu($binaryData);
	echo '<br>Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
    echo '<br>Data parsed from packet (bytes):' . PHP_EOL;
    print_r($response->getData());
	
	
} catch (Exception $exception) {
    echo '<br>An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}
```

Return:

Binary received (in hex): 01040c003700050067006f000000008d5c 
Parsed packet (in hex): 03380000000f01040c003700050067006f00000000 
Data parsed from packet (bytes): Array ( [0] => 0 [1] => 55 [2] => 0 [3] => 5 [4] => 0 [5] => 103 [6] => 0 [7] => 111 [8] => 0 [9] => 0 [10] => 0 [11] => 0 )

aldas, 2018-07-18:
couple of questions:
1. which one `->setReadTimeoutSec(5)` or `usleep(3*1000000);` helped you with reading timeout
2. have you tried to set some multiregister values ala int32 or real values in PLC and then read them with library? The thing is that different Vendors can/could have different order how bytes (words) are stored in memory. Read this http://www.simplymodbus.ca/FAQ.htm#Order 
This example how to read response as array of double words
```
    // print registers as double words in big endian low word first order (as WAGO-750 does)
    foreach ($response->getDoubleWords() as $dword) {
        print_r($dword->getInt32(Endian::BIG_ENDIAN_LOW_WORD_FIRST));
    }
```

rfsales, 2018-07-23:
During our tests both ->setReadTimeoutSec(5) and usleep(3*1000000); helped.
We´re finishing some testes and will let you know the full response.

aldas, 2018-08-20:
I'll close this. Pull request #10 added some support for RTU.

closed 2018-08-20 by aldas

vPtel1122, 2019-03-01:
First of all, thank you aldas for this great piece of work. I have downloaded and setup your modbus module in my project and it successfully connects with all the slave devices and brings in the data as well. The code using which it brings up the data is using below link code from which I just changed ip address, port, unitID, address and quantity.

https://github.com/aldas/modbus-tcp-client/blob/master/examples/index.php

Folllowing result I notice using modscan : Binary snap

[image]

Integer snap from modscan as below

[image]

Hex snap from modscan as below

[image]

Actual desired format result which I can see in the energy meter is as given in following snap from modscan using LONG INTEGER MOST SIGNIFICANT REGISTER FIRST. 

[image]

The actual data that I notice in my php web page is as below.

[image]

If we try to corelate the data between modscan and my php webpage then some similarities are found but I need the data to be as given in modscan LONG INTEGER MOST SIGNIFICANT REGISTER FIRST. I am literally clueless, I don't get it. Is there a simpler way to convert the given hex or binary to actual data that I see in the energy meter.

P.S: There may be little delay between the time snaps taken from modscan and php webpage for data corelation. The connection between php webpage and slave id exists for sure because when I input wrong ip address or wrong slave id, it throws error. The slaves are behind moxa tcp ip gateway just FYI please.

Sorry I am also not sure if this is the right place to ask for help, please correct if need be.

Any help will be greatly appreciated.

aldas, 2019-03-01:
I assume it is because of endianess/word order.  You are trying to convert 2 register to integer (2 words/32bit). Is it signed or unsigned?

Endianess is described here: 
https://github.com/aldas/modbus-tcp-client/blob/master/src/Utils/Endian.php#L44

Try this `endianess` query parameter in that example url
* http://server/example.php?endianess=1
* http://server/example.php?endianess=2
* http://server/example.php?endianess=4
* http://server/example.php?endianess=5  (this is default BIG_ENDIAN_LOW_WORD_FIRST)
* http://server/example.php?endianess=6

maybe one of those will suit. 

```
<?php

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;
use ModbusTcpClient\Utils\Endian;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('192.0.200.188')
    ->build();

$packet = new ReadHoldingRegistersRequest(197, 2);
echo 'Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;

try {
    $binaryData = $connection->connect()->sendAndReceive($packet);
    /**
     * @var $response ReadHoldingRegistersResponse
     */
    $response = (ResponseFactory::parseResponseOrThrow($binaryData))->withStartAddress(197);
    echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;

    $doubleWord = $response->getDoubleWordAt(197);
    echo 'Double word as 32 bit unsigned integer:' . PHP_EOL;
    echo 'BIG_ENDIAN: ' . $doubleWord->getUInt32(Endian::BIG_ENDIAN) . PHP_EOL;
    echo 'LITTLE_ENDIAN: ' . $doubleWord->getUInt32(Endian::LITTLE_ENDIAN) . PHP_EOL;
    echo 'BIG_ENDIAN_LOW_WORD_FIRST: ' . $doubleWord->getUInt32(Endian::BIG_ENDIAN_LOW_WORD_FIRST) . PHP_EOL;
    echo 'LITTLE_ENDIAN_LOW_WORD_FIRST: ' . $doubleWord->getUInt32(6) . PHP_EOL;
    
    echo 'Double word as 32 bit signed integer:' . PHP_EOL;
    echo 'BIG_ENDIAN: ' . $doubleWord->getInt32(Endian::BIG_ENDIAN) . PHP_EOL;
    echo 'LITTLE_ENDIAN: ' . $doubleWord->getInt32(Endian::LITTLE_ENDIAN) . PHP_EOL;
    echo 'BIG_ENDIAN_LOW_WORD_FIRST: ' . $doubleWord->getInt32(Endian::BIG_ENDIAN_LOW_WORD_FIRST) . PHP_EOL;
    echo 'LITTLE_ENDIAN_LOW_WORD_FIRST: ' . $doubleWord->getInt32(6) . PHP_EOL;
    
} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}

```

vPtel1122, 2019-03-02:
Thanks mate, your help is really appreciated. Your suggestions helped me get through. Following little piece of code did the trick.

`'$endianess = ((int)$_GET['endianess']) ?: Endian::BIG_ENDIAN;`

Though I dont get one thing that when I modscan register from 197 till length 2 then value is 1148748 at register 197 which is correct and when I scan at site register from 196 till length 2 then value is 1148748 at register 197 which is correct but the register nos went upside down, dont know how. 

Heaps of thanks mate !

aldas, 2019-03-02:
It is little bit hard without examples. could you provide example with
1. what value are you expecting, at what address (maybe picture from modscan program)
1. response example where endianess is set to BIG_ENDIAN, start address to 190 and quantity to 16

little bit wider address range for response would help me guess what is going on.

vPtel1122, 2019-03-05:
Sorry for being late to respond. The desired result is sometimes achieved at address 193, sometimes 197 depending upon at what address we start with in both modscan and php site. The snap below would explain both scenario.

[image]
[image]

Another example from address 189 from modscan and address 190 from php, qty 10 in both case. Please refer below.
[image]
[image]

Something is wrong somewhere I can't figure out. This is okay I can adjust as only one parameter I need to poll from this HPL brand energy meter. But after few days I would even need to poll several parameters from Rish Delta brand energy meter as the wire looping is going on right now. 

HPL manufacturer support team told me that KWH is receivable at address 197. Hope I have explained little better now. Thank you for your precious time.

aldas, 2019-03-05:
can you get Wireshark (https://youtu.be/_4CHW9EbIkM?t=1223) dump of packet that ModScan64 sends and receives?

I took you first example and made test code out of it. 
```
    public function testxxx()
    {
        $data = "\x89\x8a\x00\x00\x00\x17\x01\x03\x14\x00\x11\x8d\x63\x00\x00\x00\x00\x00\x00\x00\x00\x00\x01\x00\x00\x00\x32\x00\x03";

        /* @var ReadHoldingRegistersResponse $response */
        $response = ResponseFactory::parseResponse($data);

        $this->assertInstanceOf(ReadHoldingRegistersResponse::class, $response);
        $this->assertEquals(ModbusPacket::READ_HOLDING_REGISTERS, $response->getFunctionCode());

        $this->assertEquals(1150307, $response->getDoubleWordAt(0)->getUInt32(Endian::BIG_ENDIAN));
        $this->assertEquals(65536, $response->getDoubleWordAt(6)->getUInt32(Endian::BIG_ENDIAN));
        $this->assertEquals(3276803, $response->getDoubleWordAt(8)->getUInt32(Endian::BIG_ENDIAN));
    }
```

So I we query 10 registers (20 byte) and want to read first Double Register out of for unsigned 32bit integer it in BIG ENDIAN then the value for it is `00118d63` in hex

that means 2 register are combined - 2 bytes from first (0 or 196) register `\0x00\0x11` and 2 bytes for second (1 or 197) register `\0x8d\0x63`

no idea. maybe numbering of addresses if different for HIGH WORD/REGISTER FIRST.

vPtel1122, 2019-03-22:
Sorry brother ! tried playing around a lot but may be its way beyond my scope but I adjusted it with modscan actual register minus 1 equal to the register in our php class. However, I was wondering what if I want to read FC4 which is input register, how can that be achieved ? As I see under examples, there are examples for fc3,fc15 etc...is it also possible to have example for FC4 register as well ?

aldas, 2019-03-22:
FC4 is exactly the same as FC3 request see this https://github.com/aldas/modbus-tcp-client/blob/6816336b8ebe6cc7d4c4f45f794386faaa5fd3ac/src/Packet/ModbusFunction/ReadInputRegistersRequest.php#L11

so you would change https://github.com/aldas/modbus-tcp-client/blob/6816336b8ebe6cc7d4c4f45f794386faaa5fd3ac/examples/fc3.php#L17

from 
```
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity);
```
to
```
$packet = new ReadInputRegistersRequest($startAddress, $quantity);
```

vPtel1122, 2019-03-23:
Thanks mate it does work as expected. Though I had to add following as otherwise it threw fatal error of class "ReadInputRegisterRequest" not found.

`use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersResponse;`

Thank you !

moscorp, 2019-07-13:
Hi, I can run the code for several times only and I can solve it by change machine's ip address,  
it will stop on 
`$connection->connect()->send($rtuPacket);`
Possible reason : 1/serial port receive buffer overflow or 2/the network connection was refused by the server.
Is there any way to help me ?!

moscorp, 2019-07-13:
Also, it seems can't re-connect to the because of this error
`An exception occurred Unable to create client socket to tcp:http://192..`
how to fix it?!

aldas, 2019-07-13:
Could you post your code. Especially part where you create connection. From your latest post it seems that you pass 'http://192.168.x.x' as IP. It should not have 'http://' as beginning.

moscorp, 2019-07-13:
not much modified, as below:

`$ip = '192.168.1.233';

//***********************************************/

require __DIR__ . '/../vendor/autoload.php';

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;
use ModbusTcpClient\Packet\RtuConverter;

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502) //GPRS Server Port
    ->setHost($ip) //GPRS Server Address
	->setReadTimeoutSec(20) // increase read timeout to 5 seconds
    ->build();

	
$startAddress = 0; //5
$quantity = 2; //6
$slaveId = 1; // RTU packet slave id equivalent is Modbus TCP unitId

$rtuPacket = RtuConverter::toRtu(new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveId));
//$rtuPacket = new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveId);

try {
	$connection->connect()->send($rtuPacket);
        usleep(3*1000000); // sleep 0.3 seconds
        $binaryData = $connection->receive();
	$data = unpack('H*', $binaryData)[1];

	$machineid = substr($data,0,2);
	$temp = hexdec(substr($data,6,4))/10;
	$humi = hexdec(substr($data,10,4))/10;
   /*
    echo '<br>Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

    $response = RtuConverter::fromRtu($binaryData);
    echo '<br>Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
    echo '<br>Data parsed from packet (bytes):' . PHP_EOL;
    print_r($response->getData());
	*/
	
} catch (Exception $exception) {
    echo '<br>An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}`

moscorp, 2019-07-13:
I'm runing this script as cron job on every hour, It can connect and return data for 4-5hour (times) only..

moscorp, 2019-07-13:
sorry for the typo error, it is tcp://192.168.1.233:502

ShrinathGupta09, 2019-10-10:
> allright,
>
> first try increasing read timeout
>
> ```
> $connection = BinaryStreamConnection::getBuilder()
>     ->setPort(502)
>     ->setHost('127.0.0.1')
>     ->setReadTimeoutSec(5) // increase read timeout [...]
I increased it by 20 and usleep to (31000000) Fatal error: Maximum execution time of 30 seconds

aldas, 2019-10-10:
if you set ```->setReadTimeoutSec(20)``` it means that read method will try maximum 20 seconds to read response from stream.
If you execute ```usleep(31000000)``` will mean that you program will wait 31 seconds before executing next statement but seem that you PHP maximum script execution time is 30 seconds.

I would recommend playing with ```usleep();``` values.  this will make sure that device behind modbus converter has time to respond. I think having it bigger than 1000000 does not make sense.

does that RTU tcp converter work with other modbusrtu programs?

* http://www.simplymodbus.ca/RTUmaster.htm
* https://sourceforge.net/projects/easymodbustcp/