aldas/modbus-tcp-client #12: How to get value of Register?

alibaraneser, 2018-08-01:
Hi,

I want to take value of register with TCP or with RTA. How can take it ?  I think I am a little amateur in these matters, so I did not understand the exact document.

aldas, 2018-08-01:
Modbus has different functions to read registers. For example:
1. "Read Holding Registers" - http://www.simplymodbus.ca/FC03.htm
2. "Read Input Registers" - http://www.simplymodbus.ca/FC04.htm

both are pretty much the same of you consider this library. Look at this [example](examples/fc3.php)
That example: connects to Modbus at IP 127.0.0.1 and port 5020 (NB: modbus is usually at port 502 and not on ip 127.0.0.1) and sends Read holding Register request (class: ReadHoldingRegistersRequest) to fetch 6 registers starting at address 256.

if you would like to read second requested register as signed int value you would do:
```
$response = ResponseFactory::parseResponseOrThrow($binaryData);

print_r($response[1]->getInt16()); // 257th register is 1 register if request start address was 256 ( 257 - 256 = 1)

// set internal index to match start address to simplify array access
$responseWithStartAddress = $response->withStartAddress(256);

// both lines are same
print_r($responseWithStartAddress[257]->getInt16());
print_r($responseWithStartAddress->getWordAt(257)->getInt16());
```

p.s. you need to install [Composer](https://github.com/aldas/modbus-tcp-client/issues/11#issuecomment-408550486) before trying that example

alibaraneser, 2018-08-03:
Hey,

I'm getting this error message due to there is no error in Java.

Error:

Unable to create client socket to tcp://IP: Connection timed out

```
$connection = BinaryStreamConnection::getBuilder()
            ->setPort(502)
            ->setHost('IP')
            ->build();

        $startAddress = 256;
        $quantity = 6;
        $slaveId = 1;

        $packet = substr((string)new ReadHoldingRegistersRequest($startAddress, $quantity), 6);
        $packetWithCrc = Types::toByte($slaveId) . $packet . crc16($packet);

        try {
            $binaryData = $connection->connect()->sendAndReceive($packetWithCrc);
            echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

            $binaryData = substr($binaryData, 1, -2); // remove slaveId and crc
            $tcpHeader = "\x00\x00\x00\x00\x00" . Types::toByte(strlen($binaryData)) . '\x00'; // create dummy packet header
            $binaryData = $tcpHeader . $binaryData;

            $response = ResponseFactory::parseResponseOrThrow($binaryData);
            echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
            echo 'Data parsed from packet (bytes):' . PHP_EOL;
            print_r($response->getData());

        } catch (Exception $exception) {
            echo 'An exception occurred' . PHP_EOL;
            echo $exception->getMessage() . PHP_EOL;
            echo $exception->getTraceAsString() . PHP_EOL;
        } finally {
            $connection->close();
        }
```

aldas, 2018-08-10:
Are you still getting that connection error? Looking at that code you might want to change `->setHost('IP')` to real IP or domain name.

alibaraneser, 2018-08-14:
Hi,

There is no error, but results are empty array. 

ScreenShot

https://ibb.co/fUrKT9

And also, I have excel file that contain 300 register data, but your library only show less data. How can solve this problem ?

alibaraneser, 2018-08-14:
Hi, Thank for fast reply,
I'm using Siemens Climatix PLC, and I don't know starting address. How to find ?

aldas, 2018-08-14:
maybe you have different slaveId? As I understand from http://resources.flaktwoods.com/Perfion/File.aspx?id=1d3faa4b-ead8-4317-86a0-6b5b6dc61690 this plc modbus can have 2 different channels (slave/unit ids) 0/1.

Before I started this project and was investigating WAGO modbus then I actually downloaded different modbus client and tried to fetch data with them just to verify that PLC is actually working

aldas, 2018-08-14:
btw that pdf has section '4.1 Map registers' which says that different regions of register have different addreses ala 4xxxx is for "Read/Write Output or Holding Registers.
A 4x register is used to store 16-bits of numerical
data (binary or decimal), or to send the data from
the CPU to an output channel. "

alibaraneser, 2018-08-14:
Write function works, only doesn't work read data. @aldas

aldas, 2018-08-14:
this is strange. the thing is that if response does not contain any data - you see that even packet in hex is mostly zeros then it is not library but something with PLCs. maybe registers are written to one address range and should be read from different range. Try to find manual for that plc modbus block and read modbus sections - usually there is info about addressing etc

alibaraneser, 2018-08-14:
@aldas Can I send you an e-mail? , please leave this problem only for the project to end.

aldas, 2018-08-20:
I'll close this as we resolved it privately. For history sake - the problem with reading was that wrong address range was being used.

closed 2018-08-20 by aldas