aldas/modbus-tcp-client #13: Running in multithreaded mode

WeeSee, 2018-10-06:
I have some long running tasks including several Modbus requests. I want to run these tasks in a separate thread that the main process is not blocked. Using [PHP pthreads](http://php.net/manual/de/class.threaded.php) requires that every class involved in a pthread has to be derived from ```Threaded```. 

I would be cool to see this great library supporting ptreads. 

Can we expect this soon?

aldas, 2018-10-06:
and this is example of long-running process doing requests using ReactPHP https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_cli_poller.php

WeeSee, 2018-10-06:
Thanx. I did not see thes examples. Seems what I was looking for. Will try this in a few days.

closed 2018-10-06 by WeeSee