aldas/modbus-tcp-client #14: Problem with PLC

quocthanh18tn, 2018-10-10:
Hi, I admire your post. I have some problem, i try find in other forum but i can't find answer.
My problem is: when i simulate with Mod_RsSim, it's ok, but when i connect and write multi holding register, it's appear error ""An exception occurred Illegal data value "". Can you give me some solution, i try a lots but i can't fix it.
If you have any solution, please give me through email : 1513009@hcmut.edu.vn.
Thank you so much.

aldas, 2018-10-10:
could it be that you are writing to "read only" register? Have you checked your PLC manual? I know from experience from Wago PLCs that only certain address ranges are writable.

you could paste code here and maybe tell which firm/model PLC you are dealing with?

quocthanh18tn, 2018-10-12:
My code's file fc16.php. PLC i have used CPU 1214c AC/DC/RLY.  This is my server config. PLC's IP :192.168.0.2
[image]
 And data is just array [0..100] word
Connect 
[image]
And this is my code to write multi holding register 
[image] 
When i connect to PLC and run fc16.php this is my error
[image]

But when i modify fc16.php like image below  and other not modify
[image]
and this is my error :
[image]
I am final student, i dont know if my question idiot, if you have free times, can give me solution to fix it. I try to find any solution in any forum about PLC, but i cant fix it.
Sincerely thanks a lots !!

aldas, 2018-10-12:
It could be that for Function 16 it is that address '0' is not allowed.

some ideas for investigation:
1. try reading / writing data with http://www.simplymodbus.ca/TCPclient.htm client or https://sourceforge.net/projects/easymodbustcp/ - it this works then something is wrong with php lib
1. try reading data with fc3 with this lib example - if reading also fails something is really wrong
1. try to find Siemens documentation how to Modbus addresses work ala https://support.industry.siemens.com/cs/document/100633819/how-do-you-address-the-memory-areas-in-the-simatic-s7-1200-s7-1500-and-in-the-modbus-device-in-the-case-of-modbus-tcp-data-communication-?dti=0&lc=en-WW - maybe you are not allowed to write to addresses 0-4
1. check this video https://youtu.be/_4CHW9EbIkM?t=1223 - there is someone using wireshark to see what is sent in the wire 
1. maybe you have misconfigured you modbus block. Maybe looking examples as https://cache.industry.siemens.com/dl/files/340/102020340/att_118119/v3/net_modbus_tcp_s7-1500_s7-1200_en.pdf helps

so most valuable information would be to test modbus with some other client with same ports, addresses and data that is sent. At first just try reading values from modbus and then if this work try writing data. Also try to capture data packet that is sent with Wireshark and compare it with modbus packet that php lib sents

aldas, 2018-10-15:
was that siemens plc addressing document helpful to you?

quocthanh18tn, 2018-10-19:
I'm sorry for reply to you too late. I'm have a problem with some test in my college. Your code is very good, my mistake's configuration address holding register wrong number. So now, it run very well. Thank you so much. So if you have any problem that i can help you, please feel free to send me your question. :3

closed 2018-10-24 by aldas