aldas/modbus-tcp-client #15: input username and password

razvanbrasoveanu, 2018-11-06:
Hi!
First of all thank you for this library. It's awesome! 
I'm want to read data from a ET-7000/PET-7000 module, which asks for a username and password when I connect to it using the browser. I am now receiving this error:

An exception occurred Read total timeout expired

No changes if I increase the timeout period. So I'm guessing that this is caused by the authentication. The question: Can you please show me how to use authentication? Thank you!

aldas, 2018-11-06:
Hi!

Modbus protocol does not have authentication and can not have authentication. If you are connecting to PLC with browser then you are using HTTP protocol and not Modbus protocol. 

To debug this 'Read total timeout' problem I think you should check that modbus port is open. You could download Modbus client http://www.simplymodbus.ca/TCPclient.htm here and test if that port is really open. If that client can not request data then something is wrong with PLC settings.

If this does not help please let me know which example are you using and please copy/paste that example output here

razvanbrasoveanu, 2018-11-07:
> Hi!
>
> Modbus protocol does not have authentication and can not have authentication. If you are connecting to PLC with browser then you are using HTTP protocol and not Modbus protocol.
>
> To debug this [...]

Thank you! I will try and let you know.

razvanbrasoveanu, 2018-11-08:
Hello again!

I came back after the test. Using the client provided by you, i can connect to the plc and get data. Still no luck with my application. I will paste the code used bellow:
'<?php
require __DIR__ . '/vendor/autoload.php';
`<?php
require __DIR__ . '/vendor/autoload.php';

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;
use ModbusTcpClient\Utils\Endian;

// set to true if you want to let others specify their own ip/ports for querying data
// NB: this is a security risk!!!
$canChangeIpPort = false;

$ip = '192.168.132.213';
$port = 502;

if ($canChangeIpPort) {
    $ip = filter_var($_GET['ip'], FILTER_VALIDATE_IP) ? $_GET['ip'] : $ip;
    $port = ((int)$_GET['port']) ?: $port;
}

$unitId = ((int)$_GET['unitid']) ?: 0;
$address = ((int)$_GET['address']) ?: 256;
$quantity = ((int)$_GET['quantity']) ?: 12;
$endianess = ((int)$_GET['endianess']) ?: Endian::BIG_ENDIAN_LOW_WORD_FIRST;
Endian::$defaultEndian = $endianess;

echo "Using: ip: {$ip}, port: {$port}, address: {$address}, quantity: {$quantity}, endianess: {$endianess}<br>" . PHP_EOL;
$connection = BinaryStreamConnection::getBuilder()
    ->setPort($port)
    ->setHost($ip)
    ->build();

$packet = new ReadHoldingRegistersRequest($address, $quantity, $unitId);
echo 'Packet to be sent (in hex): ' . $packet->toHex() . '<br>' . PHP_EOL;

$result = [];
try {
    $binaryData = $connection->connect()->sendAndReceive($packet);
    echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . '<br>' . PHP_EOL;

    /** @var $response ReadHoldingRegistersResponse */
    $response = ResponseFactory::parseResponseOrThrow($binaryData)->withStartAddress($address);

    foreach ($response as $address => $word) {
        $doubleWord = isset($response[$address + 1]) ? $response->getDoubleWordAt($address) : null;
        $quadWord = null;
        if (isset($response[$address + 3])) {
            $quadWord = $response->getQuadWordAt($address);
            try {
                $UInt64 = $quadWord->getUInt64(); // some data can not be converted to unsigned 64bit int due PHP memory limitations
            } catch (Exception $e) {
                $UInt64 = '-';
            }
            try {
                $Int64 = $quadWord->getInt64();
            } catch (Exception $e) {
                $Int64 = '-';
            }

        }

        $highByteAsInt = $word->getHighByteAsInt();
        $lowByteAsInt = $word->getLowByteAsInt();
        $result[$address] = [
            'highByte' => '0x' . str_pad(dechex($highByteAsInt), 2, '0') . ' / ' . $highByteAsInt . ' / "&#' . $highByteAsInt . ';"',
            'lowByte' => '0x' . str_pad(dechex($lowByteAsInt), 2, '0') . ' / ' . $lowByteAsInt . ' / "&#' . $lowByteAsInt . ';"',
            'highByteBits' => sprintf('%08d', decbin($highByteAsInt)),
            'lowByteBits' => sprintf('%08d', decbin($lowByteAsInt)),
            'int16' => $word->getInt16(),
            'UInt16' => $word->getUInt16(),
            'int32' => $doubleWord ? $doubleWord->getInt32() : null,
            'UInt32' => $doubleWord ? $doubleWord->getUInt32() : null,
            'float' => $doubleWord ? $doubleWord->getFloat() : null,
            'Int64' => $quadWord ? $Int64 : null,
            'UInt64' => $quadWord ? $UInt64 : null,
        ];
    }

} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}

?>

<table border="1">
    <tr>
        <td rowspan="2">WORD<br>address</td>
        <td colspan="6">Word</td>
        <td colspan="3">Double word (from this address)</td>
        <td>Quad word</td>
    </tr>
    <tr>
        <td>high byte<br>Hex / Dec / Ascii</td>
        <td>low byte<br>Hex / Dec / Ascii</td>
        <td>high bits</td>
        <td>low bits</td>
        <td>int16</td>
        <td>UInt16</td>
        <td>int32</td>
        <td>UInt32</td>
        <td>float</td>
        <td>int64</td>
        <td>UInt64</td>
    </tr>
    <?php foreach ($result as $address => $values) { ?>
        <tr>
            <td><?php echo $address ?></td>
            <td><?php echo implode('</td><td>', $values) ?></td>
        </tr>
    <?php } ?>
</table>
Page generated: <?php echo date('c') ?>
`
And this is the result:

Using: ip: 192.168.132.213, port: 502, address: 256, quantity: 12, endianess: 5
Packet to be sent (in hex): 86f60000000600030100000c
An exception occurred Read total timeout expired #0 C:\xampp\htdocs\src\Network\BinaryStreamConnection.php(91): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom(Array, 0.3, NULL) #1 C:\xampp\htdocs\src\Network\BinaryStreamConnection.php(112): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 C:\xampp\htdocs\index.php(41): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive(Object(ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest)) #3 {main} 

If you can give me a clue on this I'll be most greatfull. Thank you!

aldas, 2018-11-08:
mmm, you could try to increase read timeout

1. increase timeout by setting custom timeout value
```
$connection = BinaryStreamConnection::getBuilder()
->setPort($port)
->setHost($ip)
->setReadTimeoutSec(5) // increase read timeout to 5 seconds
->build();
```

2. or replace this part where connection is made, packet is sent and received
```
$binaryData = $connection->connect()->sendAndReceive($packet);
```

with
```
$connection->connect()->send($packet);
usleep(3*1000000); // sleep 0.3 seconds
$binaryData = $connection->receive();
```

razvanbrasoveanu, 2018-11-09:
You are a genius! Problem solved. It was the unit parameter fault.
I changed $unitId = ((int)$_GET['unitid']) ?: 0; into $unitId = ((int)$_GET['unitid']) ?: 1;

and now it works like a charm.

Once again thank you very much for your help!

aldas, 2018-11-10:
I am glad to hear that you got it working

closed 2018-11-10 by aldas

haha8x, 2020-01-02:
I got into Read total timeout expired problem too,
I have tried all the option that u gave above but none of them work
The device I use is Moxa MGate MB3170.
- Connect using Kep Server - Working
- Connect using Modbus client - Work too

But using your code, with the same configuration that I used on the above software, it doesnt work at all. It keeps telling me Read total timeout expired. 

I tried to change unitID to diffirent number then I got Gateway path unavailable but when I change to correct number, I got Read total timeout expired again.

Can you help me find out why?

aldas, 2020-01-02:
1 .could you follow the same steps and paste/screenshot of packets that Kep Server/Modbus client is sending and what PHP is sending

> In that case I suggest sniffing traffic with Wireshark (https://www.wireshark.org/). If you are not familiar with it then maybe https://www.youtube.com/watch?v=TkCSr30UojM helps you
> 
> 1. record traffic when using Modbus client http://www.simplymodbus.ca/TCPclient.htm
> 2. record traffic when using PHP
> 

2. also you could try to add delay between sending data and reading 
change following line 
```
$binaryData = $connection->connect()
        ->sendAndReceive($packet);
```
to

```
    $connection->connect()->send($packet);
    usleep(100000); // 100ms
    $binaryData = $connection->receive();
```