aldas/modbus-tcp-client #21: Try to create buttons and form.

Cactusbonus, 2019-03-01:
Hi!
First of all thank you for this library.
I can connect my PLC (OMRON) and get data from it. The index.php from examples read memory flawlessly.
After modify the FC16.php from examples can write the PLC memory. (simply modify the value and execute)
My project is to write a very simple web based HMI with forms and buttons, but i am stuck.... I Can't link the form data to the fc16.php (  Types::toInt16(10), ) segment. I can execute the FC16.php with button but put the data in the right place only manual working for me. I am not so good for PHP programming . Can U help me?

aldas, 2019-03-02:
is that form dynamic or fields are always the same? what kind (ala integer, decimal, string) of form fields needs to be written with FC16?

Cactusbonus, 2019-03-03:
Hi!
I would like to create one "read" /"write" PLC page with Read memory areas and some input field (integer) and a modify and  BIT switch (ON/OFF) button. 
Something like this [https://static1.squarespace.com/static/59dabfa0ccc5c54f48d42627/t/5a0b47d6ec212dd299c10d79/1510688732946/GreenHighlight_Box_conveyors.JPG](url) but i need to write memory area with input field with a modify button to change speed.
Thx for help.

aldas, 2019-03-03:
okey, if you are able to create [html form](https://www.w3schools.com/php/php_forms.asp)  and send 'converter 1 speed' as `conv1_speed` form field then on PHP side you can act on that this way

form.html
```
<html>
<body>

<form action="post.php" method="post">
Converter 1 speed: <input type="text" name="conv1_speed"><br>
<input type="submit">
</form>

</body>
</html>
```

post.php
```
// only write when input variable was in POST request and value sent validates to an integer
if (isset($_POST['conv1_speed']) && filter_var($_POST['conv1_speed'], FILTER_VALIDATE_INT) !== FALSE) {
    $conv1_speed = (int)$_POST['conv1_speed'];

    $startAddress = 12288;
    $registers = [Types::toInt16($conv1_speed)];
    $packet = new WriteMultipleRegistersRequest($startAddress, $registers);
}
```

Cactusbonus, 2019-03-04:
Hi!
Your code working flawlessly Thx!!! Only one thing.. When I press the submit button the form.html the page jump to the post.php and not stay in form.html. Can this solving somehow? Thx....

aldas, 2019-03-04:
To achieve that your `post.php` should redirect browser back to `form.html` after data is written to modbus.

Redirect is done by sending ["location"](https://stackoverflow.com/a/768472) header after data is sent

```
$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->build();

// only write when input variable was in POST request and value sent validates to an integer
if (isset($_POST['conv1_speed']) && filter_var($_POST['conv1_speed'], FILTER_VALIDATE_INT) !== FALSE) {
    $conv1_speed = (int)$_POST['conv1_speed'];

    $startAddress = 12288; // conv1_speed modbus address
    $registers = [Types::toInt16($conv1_speed)];
    $packet = new WriteMultipleRegistersRequest($startAddress, $registers);

    try {
        $binaryData = $connection->connect()->sendAndReceive($packet);

        // redirect poster back to form
        header('Location: form.html', 301);
    } catch (Exception $exception) {
        echo 'An exception occurred' . PHP_EOL;
        echo $exception->getMessage() . PHP_EOL;
        echo $exception->getTraceAsString() . PHP_EOL;
    } finally {
        $connection->close();
        exit();
    }
}
```

Cactusbonus, 2019-03-05:
Hi! Thx for the exellent redirection code, working well. If i am not too annoying... :)  Can I ask u another question? The code working well when creating a table and filling it as many data as i want. But when i put the some javacript code (animated gauge) to read memory places only work when "$quantity = 1" when i want more than one data from my PLC only the last data wirte to the javascript data place. For example: the original code was: arcMeter( '#myArc1',0.33, options );   the new code arcMeter( '#myArc1',<?php echo implode ($values)?>, options ); so i put the <?php echo implode ($values)?> instead of 0.33 and the result is if the $quantity more then 1 the last data wite to the code. How can use multiple gauge/meter how can read separete data from PLC? I tested is some  javascript gauge always same result.  
Thx

aldas, 2019-03-05:
I assume you are using https://github.com/dleber/arcMeter it take single value. Implode is for different things http://php.net/manual/en/function.implode.php

Lets assume that you are using that `index.php` as base. Then `$value` is an assoc array that contains different data type names as keys. To create arcMeter for  int16 you would do

```
arcMeter( '#myArc1',<?php echo $values['int16'] ?>, options );
```
Read this https://www.w3schools.com/php/php_arrays.asp

Cactusbonus, 2019-03-06:
Thx for the answer. Unfortunetly after i using `arcMeter( '#myArc1',<?php echo $values['int16'] ?>, options );` same result. When the array had more than 1 data the last data written the meter and i can't separete the data call from the  `$values` array. I found a less elegant solution if no other options. Every meter get a full code with different variables and arrays. Another unfortunetly thing is the OMRON FBD only let one connection/time so i place a sleep time between the connecting codes. BTW thx a lot... you save me...

closed 2019-03-06 by aldas