aldas/modbus-tcp-client #23: Parsing data based on endian

harli91, 2019-03-13:
How do i pass the endinan to the parse function in a tcp connection?

`$request->parse($data);`

`$request->parse($data,$endian);`

aldas, 2019-03-13:
With that
https://github.com/aldas/modbus-tcp-client/blob/6816336b8ebe6cc7d4c4f45f794386faaa5fd3ac/examples/index.php#L25

harli91, 2019-03-14:
i have set the endianess like this

```
$endianness = [
        "ABCD" => Endian::BIG_ENDIAN,
        "CDAB" => Endian::BIG_ENDIAN_LOW_WORD_FIRST,
        "DCBA" => Endian::LITTLE_ENDIAN,
        "BADC" => Endian::BIG_ENDIAN_LOW_WORD_FIRST,
        "AB"   => Endian::BIG_ENDIAN,
        "BA"   => Endian::LITTLE_ENDIAN
    ];
```
is this right?

aldas, 2019-03-14:
yep, although ABCD and AB is the same. Word order is only meaningful if you are dealing data types that are bigger than 1 word/register. So for (u)int16 it does not matter but anything related to float/int32/int64 it is important.

harli91, 2019-03-15:
thanks man, grate library :)

closed 2019-03-15 by harli91