aldas/modbus-tcp-client #24: New to modbus

jamenlang, 2019-03-20:
The project says this is a modbus client. Can I use it as a PLC emulator? I'd like to be able to poll data with SNMP and convert it to modbus acting as a PLC would.

server -> php:
wait for modbus request
-> get one, figure out what device it's for,
-> poll the device for snmp data.
-> convert snmp data to modbus coils registers... idk
-> return response.

aldas, 2019-03-20:
Well, you could use this library to respond modbus packets. I'm using this in tests see https://github.com/aldas/modbus-tcp-client/blob/master/tests/integration/Fc3ReadHoldingRegistersTest.php#L16

and server to respond data https://github.com/aldas/modbus-tcp-client/blob/6816336b8ebe6cc7d4c4f45f794386faaa5fd3ac/tests/integration/MockResponseServer.php#L45

so you could change ```pack('H*', $responsePacket)``` with ```(new ReadCoilsResponse($data))->__toString()```

jamenlang, 2019-03-20:
Do i need react for that?

aldas, 2019-03-20:
well by default php is used to serve as http server but to act as a modbus server, you need something that runs your PHP as plain tcp"socket" server and accepts connections to which your code responds data (modbus packets).  Instead of React you could use https://amphp.org/socket/ or roll your own server.

jamenlang, 2019-03-20:
i understand now, 

i installed amphp with composer i think.

composer require amphp/socket, i'm guessing react is installed the same way.

aldas, 2019-03-20:
yep, see this https://github.com/reactphp/socket

jamenlang, 2019-03-20:
looks like i'll have to install react datagram as well?

composer require react/datagram:^1.4

aldas, 2019-03-20:
React Datagram (https://github.com/reactphp/datagram) is needed when you want to build modbus UDP server. If you want serve your requests over TCP you only need https://github.com/reactphp/socket

closed 2019-03-25 by aldas