aldas/modbus-tcp-client #30: example_cli_poller.php not closing connection

micbuh, 2019-04-16:
I am testing 'example_cli_poller.php' against [modrsim](http://www.plcsimulator.org/). By default, the simulator allows 10 concurrent connections. But, with every poll, a new connection is used. The 11st attempt fails. It seems like connections are not properly closed. 

> $ netstat | grep :502
tcp        0      0 10.0.2.15:38354         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38360         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38364         192.168.100.98:502      ESTABLISHED
tcp        0      0 10.0.2.15:38350         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38362         192.168.100.98:502      ESTABLISHED
tcp        0      0 10.0.2.15:38358         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38356         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38352         192.168.100.98:502      FIN_WAIT2  
tcp        0      0 10.0.2.15:38352         192.168.100.98:502      FIN_WAIT2

'example_parallel_requests.php' shows a similar behaviour. 

If other devices connect to the simulator, only one connection is used while polling cyclic.
Allthough, the connection simply might be kept open in this case.

Is there an easy fix? How to properly poll once per second?

Thanks, Michael

PS. Great library and concept!

aldas, 2019-04-17:
I have noticed that FIN_WAIT stuff also, so far for WAGO controller this have not been an issue. I you want to limit connections you could open connection or N number of connections out of polling loop and use them in loop (as connection pool). 

'example_cli_poller.php' is using React Socket (https://github.com/reactphp/socket) library to open/close connection and is done far more knowledgeable guys/gals than me - if they have not done it right it may be some PHP limitation - maybe socket that is waiting to be closed can really be closed/recycled when PHP script ends. Maybe https://www.php.net/manual/en/function.stream-socket-shutdown.php would help.

aldas, 2019-04-19:
@micbuh maybe you could try to add ```stream_socket_shutdown($this->stream,STREAM_SHUT_RDWR);``` near https://github.com/aldas/modbus-tcp-client/blob/e22a138d5d555c01600322cd3eee1794d0d96fb9/src/Network/BinaryStreamConnection.php#L118 and see if it helps

closed 2019-05-18 by aldas