aldas/modbus-tcp-client #31: Problem with ReadHoldingRegistersRequest

bilk, 2019-04-24:
I am working with Modbus TCP protocol client, and I ended up in a problem that I do not know the same cause, when I pass an array through function 16 with WriteMultipleRegistersRequest `$registers = [Types::toByte(10), Types::toByte(10),];` and call the function 3 ReadHoldingRegistersRequest I have the following return `Data parsed from packet (bytes): " [9]=> array(10) { [0]=> int(0) [1]=> int(10) [2]=> int(0) [3]=> int(10) [4]=> int(0) [5]=> int(12) [6]=> int(0) [7]=> int(0) [8]=> int(0) [9]=> int(3) } `.
Then the questioning, if I have moved from position 0 the values (10, 10), the positions 0 and 1 of the retriever array would not have to return the same values ?, or am I making a mistake ?

aldas, 2019-04-24:
Could you post your code? and output of something similar
```
$connection = (BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->build())->connect();

$packet = new WriteMultipleRegistersRequest(256, [Types::toByte(10), Types::toByte(10)]);
echo 'FC16 Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;
$binaryData = $connection->sendAndReceive($packet);
echo 'FC 16 Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

$packet = new ReadHoldingRegistersRequest(256, 2);
echo 'FC3 Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;
$binaryData = $connection->sendAndReceive($packet);
$response = ResponseFactory::parseResponseOrThrow($binaryData);
echo 'FC3 Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
```

bilk, 2019-04-24:
`$connection = BinaryStreamConnection::getBuilder()
            ->setPort(502)
            ->setHost('192.168.25.100')
            ->build();

$startAddress = 200;
        
$registers = [
       Types::toByte(10),
       Types::toByte(10)
];
$packet = new WriteMultipleRegistersRequest($startAddress, $registers);

echo 'Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;

$binaryData = $connection->connect()->sendAndReceive($packet);

echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

$response = ResponseFactory::parseResponseOrThrow($binaryData);
echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
echo 'Registers count written parsed from packet:' . PHP_EOL;
echo $response->getRegistersCount();
            
$packet = new ReadHoldingRegistersRequest(200, 5);
echo 'Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;
            
$binaryData = $connection->connect()->sendAndReceive($packet);
echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;
 
$response = ResponseFactory::parseResponseOrThrow($binaryData);
echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
echo 'Data parsed from packet (bytes):' . PHP_EOL;
echo $response->getData();`

aldas, 2019-04-24:
Okey, I see. `$response->getData()` returns raw data. this is 10 bytes for 5 registers. (register is 2 bytes)

so array indexes from 0 to 3 are bytes for your registers that you wrote with fc16
```
[0]=> int(0) 
[1]=> int(10) 
[2]=> int(0) 
[3]=> int(10) 
[4]=> int(0) 
[5]=> int(12) 
[6]=> int(0) 
[7]=> int(0) 
[8]=> int(0) 
[9]=> int(3)
```

If you want to read specific register values do this
```
print_r($response[0]->getInt16()); // use array access to get register
print_r($response[0]->getBytes()); // use array access to get register
print_r($response[1]->getInt16()); // use array access to get register
print_r($response[1]->getBytes()); // use array access to get register

// or loop over returned registers
foreach ($response as $register) {
     /** @var $register Word */
    print_r($register->getInt16());
}

// or
// set internal index to match start address to simplify array access
$responseWithStartAddress = $response->withStartAddress($startAddress);
print_r($responseWithStartAddress[$startAddress]->getInt16());
```

bilk, 2019-04-24:
That's exactly what I needed, thank you!

closed 2019-04-26 by aldas