aldas/modbus-tcp-client #33: Register 0 not accepted.

Nemecsek, 2019-07-11:
This could be related to #28.

I am reading a Janitza UMG604.
It provides a register 0 but I cannot read it (I can request any register > 0):

```
Endian::$defaultEndian = Endian::BIG_ENDIAN;

$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://X.X.X.X', 
    ->int64(0, "realtime")
    ->int32(4, "sys time")
    ->build();
```

The error is `Address out of bounds` and comes from `ReadHoldingRegistersResponse.php`, line 180:
```
/**
     * @param $firstWordAddress
     * @return QuadWord
     */
    public function getQuadWordAt(int $firstWordAddress)
    {
        $address = ($firstWordAddress - $this->getStartAddress()) * 2;
        $byteCount = $this->getByteCount();
        if ($address < 0 || ($address + 8) > $byteCount) {
            throw new InvalidArgumentException('address out of bounds');
        }
        return new QuadWord(substr($this->data, $address, 8));
    }
```
The error happens if we read more than 1 register. These are some variables in `getQuadWordAt` according to my tests. It looks like `$byteCount` is wrong but I couldn't find the issue:

```
// SINGLE REGISTER 0: OK
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
    ->int64( 0, "realtime")
// getStartAddress 0, address 0, bytecount 8, firstWordAddress 0

// 2 REGISTERS FROM 0: ERROR
// the size of first register has been lost!
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
    ->int64( 0, "realtime")
    ->int32( 4, "sys time")
// getStartAddress 4, address -8, bytecount 4, firstWordAddress 0

// 3 REGISTERS FROM 0: ERROR
// the size of first register has been lost!
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
    ->int64( 0, "realtime")
    ->int32( 4, "sys time")
    ->int16( 6, "day")
// getStartAddress 4, address -8, bytecount 6, firstWordAddress 0

// 4 REGISTERS FROM 0: ERROR
// the size of first register has been lost!
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
    ->int64( 0, "realtime")
    ->int32( 4, "sys time")
    ->int16( 6, "day")
    ->int16( 7, "month")
// getStartAddress 4, address -8, bytecount 8, firstWordAddress 0

// 3 REGISTERS FROM 4: OK
// faulty register has been commented out
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
//  ->int64( 0, "realtime")  
    ->int32( 4, "sys time")
    ->int16( 6, "day")
    ->int16( 7, "month")
// getStartAddress 4, address 0, bytecount 8, firstWordAddress 4

// 3 REGISTERS FROM 0, WITH HOLE: ERROR
// the first register has been lost; the error is obviously not related to int64 alone
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://10.10.3.2:502', 1)
    ->int32( 0, "dummy 32bit register") 
    ->int16( 6, "day")
    ->int16( 7, "month")
// getStartAddress 6, address -12, bytecount 4, firstWordAddress 0
```

aldas, 2019-07-11:
Hi! thanks reporting the problem. your described problem is in method where addresses are splitted into separate requests.

https://github.com/aldas/modbus-tcp-client/commit/4157cab728abb2ffe63ffe0e684f57167f1b140b#diff-f5ead4071f50c5dfafce5c6459c3ac7dR50

I have made a bug fix. could you checkout master/latest changes see if it helps.

Nemecsek, 2019-07-12:
It works! 
Problem has been solved. Very fast, thank you.

commit 7b344f78 (2019-07-11): fixes #33 - ReadRegistersBuilder address splitter calculates start address incorrectly when value is 0.

also fixes case when addresses overlap then quantity was calculated incorrectly
commit 46e41f64 (2019-07-11): fixes #33 - ReadRegistersBuilder address splitter fixes

1. it calculates start address incorrectly when value is 0.
2. also fixes case when addresses overlap then quantity was calculated incorrectly
PR #35 (2019-07-11): fixes #33 - ReadRegistersBuilder address splitter calculates start ad…
* ReadRegistersBuilder address splitter calculates start address incorrectly when value is 0.
* also fixes case when addresses overlap then quantity was calculated incorrectly

fixes #33
commit f5d93461 (2019-07-11): fixes #33 - ReadRegistersBuilder address splitter fixes

1. it calculates start address incorrectly when value is 0.
2. also fixes case when addresses overlap then quantity was calculated incorrectly
closed 2019-07-12 by Nemecsek