aldas/modbus-tcp-client #37: Call to undefined method ReadCoilsResponse::getWordAt() / ReadInputDiscretesResponse::getWordAt()

harli91, 2019-07-16:
When i try to get some values using FC1 or FC2 and `allFromArray()`  function i get the following errors at parsing

`Call to undefined method ModbusTcpClient\Packet\ModbusFunction\ReadCoilsResponse::getWordAt()`

`Call to undefined method ModbusTcpClient\Packet\ModbusFunction\ReadInputDiscretesResponse::getWordAt()`

aldas, 2019-07-16:
@harli91 could you paste sample of you code?

`allFromArray` is method of `ReadRegistersBuilder` and `WriteRegistersBuilder` and those classes are used to make FC3/FC4 requests.

Are you creating your own instance of `ReadRegistersBuilder` instance with default constructor? ala 

```
$x = new ReadRegistersBuilder(ReadCoilsRequest::class, $uri, $unitId);
```

actually `ReadRegistersBuilder` *Registers* in name says that it mean to for register type requests but this does not mean that we could not have `ReadCoilsBuilder` like class what does what you need.

anyway, could you show me your code/usecase and maybe I help you

harli91, 2019-07-17:
As you sad i build the request using an new instance of ReadRegistersBuilder

```
$fncArray = [
		"3"=>"newReadHoldingRegisters",
		"4"=>"newReadInputRegisters",
		"2" => ModbusTcpClient\Packet\ModbusFunction\ReadInputDiscretesRequest::class,
                "1" => ModbusTcpClient\Packet\ModbusFunction\ReadCoilsRequest::class
		];
		if(in_array($fnc,["1","2"])) {
			$rq = new ModbusTcpClient\Composer\Read\ReadRegistersBuilder($fncArray[$fnc]);
		} else {
			$rq = ModbusTcpClient\Composer\Read\ReadRegistersBuilder::{$fncArray[$fnc]}();
		}
		$rq = $rq->allFromArray($data)
```
and when i parse using 

`$parsedResponse = $this->request->parse($data); ` i get the errors

The errors i think originate  `ModbusTcpClient\Composer\Read\ReadAddress`, function `extractInternal`

aldas, 2019-07-17:
Coils/InputDiscrete (fc1/fc2) and (Holding/Input)Register (fc3/fc4) or 2 different things.  Coil is boolean - it is 1 (on) or 0 (off)  but  Register (aka Word) is 2 bytes of any data - what data type it is your to decide - it can be readout temperature sensor. 

What `ReadRegistersBuilder` does, it helps you to retrieve data in a data type that you need. It does not make sense for me to fetch array of bits (boolean) and combine 16 coils to get 1 int16 value. 

could you `print_r` maybe value of `$data`  for line `$rq = $rq->allFromArray($data)` I would like to see what aadresses and data types are this code fetching with fc1/fc2. Are you fetching only `bit` values when requesting fc1/fc2?

harli91, 2019-07-19:
if i use `bit` or `byte` i get a similar error in  the same `extractInternal` function in classes `ModbusTcpClient\Composer\Address\BitReadAddress(ByteReadAddress)` 

here is a print_r of a byte request

Array
```
(
    [0] => ModbusTcpClient\Composer\Read\ReadRequest Object
        (
            [uri:ModbusTcpClient\Composer\Read\ReadRequest:private] => tcp://192.168.1.103:502
            [addresses:ModbusTcpClient\Composer\Read\ReadRequest:private] => Array
                (
                    [0] => ModbusTcpClient\Composer\Read\ByteReadAddress Object
                        (
                            [firstByte:ModbusTcpClient\Composer\Read\ByteReadAddress:private] => 1
                            [name:ModbusTcpClient\Composer\Read\ReadAddress:private] => test
                            [callback:protected] => 
                            [errorCallback:ModbusTcpClient\Composer\Read\ReadAddress:private] => 
                            [address:protected] => 5
                            [type:protected] => byte
                        )

                )

            [request:ModbusTcpClient\Composer\Read\ReadRequest:private] => ModbusTcpClient\Packet\ModbusFunction\ReadCoilsRequest Object
                (
                    [quantity:ModbusTcpClient\Packet\ModbusFunction\ReadCoilsRequest:private] => 1
                    [startAddress:ModbusTcpClient\Packet\ProtocolDataUnitRequest:private] => 5
                    [header:ModbusTcpClient\Packet\ProtocolDataUnit:private] => ModbusTcpClient\Packet\ModbusApplicationHeader Object
                        (
                            [transactionId:ModbusTcpClient\Packet\ModbusApplicationHeader:private] => 10682
                            [length:ModbusTcpClient\Packet\ModbusApplicationHeader:private] => 6
                            [unitId:ModbusTcpClient\Packet\ModbusApplicationHeader:private] => 0
                        )

                )

        )

)
```

aldas, 2019-07-21:
still `ReadRegistersBuilder` is not meant for FC1/FC2 and should not be used with FC1/FC2 as it only supports registers and not coils as addresses. 

I created #38 to deal with request builder for modbus coils (fc1/fc2)

closed 2019-07-21 by aldas