aldas/modbus-tcp-client #39: Unable to use Uint on F6 due validate() validates only int16.

rfsales, 2019-09-17:
We are trying to set value '65534' on a Uint16 address but an exception occurred:

Uncaught ModbusTcpClient\Exception\InvalidArgumentException: value is not set or out of range (int16): 65534 in .../aldas/modbus-tcp-client/src/Packet/ModbusFunction/WriteSingleRegisterRequest.php:38

Function 6, file: src/Packet/ModbusFunction/WriteSingleRegisterRequest.php

public function validate()
    {
        parent::validate();
        if ((null !== $this->value) && (($this->value >= Types::MIN_VALUE_INT16) && ($this->value <= Types::MAX_VALUE_INT16))) {
            return;
        }
        throw new InvalidArgumentException("value is not set or out of range (int16): {$this->value}");
    }

closed 2019-09-17 by aldas

aldas, 2019-09-17:
fixed. created new bugfix version 1.1.2

rfsales, 2019-09-18:
Aldas as always solving the world problems!!!
Thanks a lot!

PR #40 (2019-09-17): fix #39 Write Single Register (FC=06) value validation not allowing u…
fix #39 Write Single Register (FC=06) value validation not allowing usigned int16 values to be used
commit 3d0076e6 (2019-09-17): fix #39 Write Single Register (FC=06) value validation not allowing usigned int16 values to be used