aldas/modbus-tcp-client #41: autoload.php

piotrw-git, 2019-09-18:
Hello,
What does the autoload.php file contain?

aldas, 2019-09-18:
Do you mean file `vendor/autoload.php`? This is composer autoloading file see `https://getcomposer.org/doc/`

this library is meant to be used with `composer` https://getcomposer.org/

closed 2019-09-25 by aldas

piotrw-git, 2019-10-03:
Hello,

Sorry, I still don't understand what this file should contain?

Which files are required additionally?

I have always used require_once ()

aldas, 2019-10-03:
I strongly suggest that you read what composer is. Please see

- https://www.youtube.com/watch?v=I6wm15OWyqg
- https://www.youtube.com/watch?v=d_zPOJ8cz_A
- https://www.codewall.co.uk/how-to-install-use-composer-tutorial-for-dummies-with-pdf/

this will help you in the future