aldas/modbus-tcp-client #42: An exception occurred Response null or data length too short to be valid packet!

ShrinathGupta09, 2019-10-02:
Hi There 
Thanks for the library and I am new to Modbus protocol so this question might be naive. I used the library with Modbus Slave Software and it worked flawlessly but when I connect my laptop to actually devices an exception is thrown response packet to be null.
I have 2 devices connected to the RS485 to Modbus TCP Converter and the setting for the devices are 
Baud Rate 9600 bits/second
Data Size 8 bits/character
Parity Odd
Port Number502
Network Protocol  TCP
RS485 to Modbus TCP Convertor IP is 169.254.205.239
And here is my code

<?php
include 'database.php';
use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;
require __DIR__ . '/vendor/autoload.php';
//$server_ip = getHostByName(getHostName());
$server_ip = '169.254.205.239';
//echo "Server IP Address is: $server_ip"; 
$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost($server_ip)
    ->build();
$startAddress = 40043;
$quantity = 10;
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity);
//echo 'Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;
try {
    $binaryData = $connection->connect()
        ->sendAndReceive($packet);
    //echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;
    /**
     * @var $response ReadHoldingRegistersResponse
     */
    $response = ResponseFactory::parseResponseOrThrow($binaryData);
  //  echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
  //  echo 'Data parsed from packet (bytes):' . PHP_EOL;
   print_r($response->getData());
	$array = $response->getData();
	$query = "SELECT * FROM  holdingregister WHERE id='1' ;";
	mysqli_query($con,$query);
	$count = mysqli_affected_rows($con);
	if($count>0) {
		$sql = "UPDATE holdingregister SET register1 ='$array[1]' , register2 = '$array[3]' , register3 = '$array[5]' WHERE id = 1";
			mysqli_query($con,$sql);
	}
		else{
			$sql = "INSERT INTO holdingregister(id,register1,register2,register3) VALUES ('1',$array[1],$array[3],$array[5])";
			mysqli_query($con,$sql);
		}
			
   /* foreach ($response as $word) {
        print_r($word->getBytes());
    }
    foreach ($response->asDoubleWords() as $doubleWord) {
        print_r($doubleWord->getBytes());
    }
    // set internal index to match start address to simplify array access
    $responseWithStartAddress = $response->withStartAddress($startAddress);
    print_r($responseWithStartAddress[5000]->getBytes()); // use array access to get word
    print_r($responseWithStartAddress->getDoubleWordAt(5001)->getFloat());*/
} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}

aldas, 2019-10-02:
1. by default unitID/SlaveId is 0. Maybe you need to use different slaveID ala 1
```
slaveID = 1;
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveID);
```

2. maybe you need to wait a little before reading
change from using ```sendAndReceive``` to
```
    $connection->connect()->send($packet);
    usleep(100000); // wait 100ms
    $binaryData = $connection->receive();
```

3. use wireshark to see what `Modbus Slave Software` sends and compare it to what is library sends.

ShrinathGupta09, 2019-10-08:
> 
> 
>     1. by default unitID/SlaveId is 0. Maybe you need to use different slaveID ala 1
> 
> 
> ```
> $slaveID = 1;
> $packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveID);
> ```
> 
>     1. maybe you need to wait a little before reading
>        change from using `sendAndReceive` to
> 
> 
> ```
>     $connection->connect()->send($packet);
>     usleep(100000); // wait 100ms
>     $binaryData = $connection->receive();
> ```
> 
>     1. use wireshark to see what `Modbus Slave Software` sends and compare it to what is library sends.

I tried the solutions you gave but it didn't work. Actually, the data packet sent is fine but the response packet is null for the reasons i don't know

ShrinathGupta09, 2019-10-08:
Packet to be sent (in hex): ad8500000006010300040002 Binary received (in hex): 00 An exception occurred Packet crc (\x00) does not match calculated crc (\xffff)! #0 C:\xampp\htdocs\modbus\readRegister.php(30): ModbusTcpClient\Packet\RtuConverter::fromRtu('\x00') #1 C:\xampp\htdocs\modbus\HRLoad.php(13): include('C:\\xampp\\htdocs...') #2 {main}

aldas, 2019-10-08:
maybe you could try with some other client 
ala 
* http://www.simplymodbus.ca/RTUmaster.htm 
* https://sourceforge.net/projects/easymodbustcp/

just to see if it works at all.

ShrinathGupta09, 2019-10-10:
Earlier I did a mistake I made the packet for modbus tcp while I was sending it to the modbus rtu devices.I saw the issue 'This project supports Modbus RTU'  and i am facing readtimeout even on increased readtimeout sec I have posted the reply in the closed issue please kindly do take a look

closed 2019-10-17 by aldas