aldas/modbus-tcp-client #43: Read multiple packets with low level

marioshtika, 2020-01-08:
Is there a way to read multiple packets using the low-level script?

I have more than 100 addresses from the same Uri and it is taking a lot of time to finish all the requests.
As I understand for every request, it connects, gets value, closes the connection. And this will happen for every request (more than 100 times).

aldas, 2020-01-08:
If it is OK to do them serially then maybe this would help

```
<?php

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ResponseFactory;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->build();

$result = [];
$requests = [
    ['address' => 256, 'quantity' => 6],
    ['address' => 270, 'quantity' => 100],
];

try {
    $connection->connect();

    foreach ($requests as $request) {
        $address = $request['address'];
        $quantity = $request['quantity'];

        $binaryData = $connection->sendAndReceive(new ReadHoldingRegistersRequest($address, $quantity));

        $response = ResponseFactory::parseResponseOrThrow($binaryData);
        echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
        $result[$address . '_' . $quantity] = $response->getData();
    }

    print_r($result);
} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}
```

aldas, 2020-01-08:
this will solve only opening/closing connection part. If you need to do multiple requests in 'parallel' then see 'examples/example_parallel_requests_reactphp.php' and instead of using `ReadRegistersBuilder::newReadHoldingRegisters` create array of requests by youself and use them in `foreach ($requests as $request) {` loop

there is also example for `example/example_parallel_requests_amp.php`

marioshtika, 2020-01-09:
Hello @aldas and thank you very much for your response and your help.

I followed your advice and the time was reduced by 2 seconds. 
(Every improvement is a good improvement 😎)

marioshtika, 2020-01-09:
I was using the parallel request with amp before (and it is working way faster) but I was having a problem with **Connection timeout exceeding on multiple requests** by getting the below error message

```
Exception 'Amp\Socket\ConnectException' with message 'Connecting to tcp://192.168.0.220:502 failed: timeout exceeded (10000 ms)'

in /vagrant/vendor/amphp/socket/src/DnsConnector.php:82
```
_By the way, I am using a Dixell XR77CH device_

Am I doing something wrong or is this a limitation for the device?

I also opened an issue to the socket amp repository
https://github.com/amphp/socket/issues/68

and they suggested (if it is possible) not to make a connection for each packet.

Is there a way to solve the opening/closing connection part for parallel requests as we did above?
Or do you have any other suggestions?

Thank you in advance.

aldas, 2020-01-09:
I think you need to build some kind of "connection pool" - so only X amount of request can be ongoing at certain time. 

From my own experience with Wago controllers I know that PLCs could have limited amount of sockets that modbus can use/open. so for our case we were limited/capped to 10 simultaneous connections to PLC.

marioshtika, 2020-01-13:
Thank you @aldas, you helped me a lot.

closed 2020-01-13 by marioshtika