aldas/modbus-tcp-client #44: Extend the maximum read time of a request in higher level

marioshtika, 2020-01-16:
Hello @aldas ,

I am using your example on [parallel request with amp](https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_parallel_requests_amp.php). 
I am trying to simultaneously connect and read multiple units.

Some of the request are giving me the below error
```
Connecting to tcp://192.168.0.223:502 failed: timeout exceeded (10000 ms)
```

In the low level I am using the `->setReadTimeoutSec(20)` command.
Is there a way to do the same in the higher level?

Thank you in advance.

aldas, 2020-01-18:
You can set Context for connect() method

```
            $connectContext = (new ConnectContext)->withConnectTimeout(1000); // 1sec
            /** @var \Amp\Socket\ClientSocket $socket */
            $socket = yield connect($request->getUri(), $connectContext);
```

but connect timeout and read timeout are different things.

aldas, 2020-01-25:
Closing this ticket atm. PhpAmp library related questions are better to be asked at their repo.

closed 2020-01-25 by aldas