aldas/modbus-tcp-client #47: Implement function 23 on modbus

rfsales, 2020-03-27:
Hello Aldas!
It´s me again :)

There is any to use function 23 on this library?
I got a new cenario and f23 will reduce a lot of the communication.

aldas, 2020-03-27:
Currently there is no F23 implementation (Read/Write Multiple registers) but it could be added.

For history sake i'll paste modbus spec info about it here:
http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf
> 6.17 23 (0x17) Read/Write Multiple registers .................................................................38
> This function code performs a combination of one read operation and one write operation in a
single MODBUS transaction. The write operation is performed before the read.
Holding registers are addressed starting at zero. Therefore holding registers 1-16 are
addressed in the PDU as 0-15. The request specifies the starting address and number of holding registers to be read as well as the starting address, number of holding registers, and the data to be written. The byte count specifies the number of bytes to follow in the write data field. The normal response contains the data from the group of registers that were read. The byte
count field specifies the quantity of bytes to follow in the read data field.

it is almost combination of WriteMultipleRegistersRequest and ReadHoldingRegistersResponse - not that hard to create.

aldas, 2020-03-27:
@rfsales could you try "fc23_implementation" #48  branch out and if it works for you i'll merge it into master and tag new version of library.

rfsales, 2020-03-28:
That's great! Yes for sure i'll test it next week! 
Did you include any example on the doc area? 
Thank you a lot!

aldas, 2020-03-28:
Example can be found here https://github.com/aldas/modbus-tcp-client/blob/fc23_implementation/examples/fc23.php

PR #48 (2020-04-05): Fc23 implementation
resolves #47

* adds FC23 function implementation 
* adds request time on example request page
closed 2020-04-05 by aldas