aldas/modbus-tcp-client #55: Implement missing (serial) functions (7,8,11,12,17,20,21,22,24,43)

aldas, 2020-04-12:
Implement missing serial functions (7,8,11,12,17,20,21,24,43)

Modbus spec: http://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

* 07 (0x07) Read Exception Status (Serial Line only) 
* 08 (0x08) Diagnostics (Serial Line only)
* 11 (0x0B) Get Comm Event Counter (Serial Line only) 
* 12 (0x0C) Get Comm Event Log (Serial Line only) 
* 17 (0x11) Report Server ID (Serial Line only) 
* 20 (0x14) Read File Record
* 21 (0x15) Write File Record 
* 22 (0x16) Mask Write Register 
* 24 (0x18) Read FIFO Queue 
* 43 ( 0x2B) Encapsulated Interface Transport 
  *  43 / 13 (0x2B / 0x0D) CANopen General Reference Request and Response
PDU
  * 43 / 14 (0x2B / 0x0E) Read Device Identification 

If anyone needs some of them let me know

thanhnvh01, 2023-12-19:
i need 17 (0x11)

aldas, 2024-01-01:
This may not be relevant anymore for you @thanhnvh01 but FC17 is implemented now: [ReportServerIDRequest](src/Packet/ModbusFunction/ReportServerIDRequest.php) / [ReportServerIDResponse](src/Packet/ModbusFunction/ReportServerIDResponse.php)

thanhnvh01, 2024-01-05:
> This may not be relevant anymore for you @thanhnvh01 but FC17 is implemented now: [ReportServerIDRequest](https://github.com/aldas/modbus-tcp-client/tree/master/src/Packet/ModbusFunction/ReportServerIDRequest.php) / [ReportServerIDResponse](https://github.com/aldas/modbus-tcp-client/tree/master/src/Packet/ModbusFunction/ReportServerIDResponse.php)

Can you do FC17 with go https://github.com/aldas/go-modbus-client.git
I want scan slave id but i dont find any library in golang to do this

aldas, 2024-01-05:
Yes, I can add FC there.

aldas, 2024-01-07:
@thanhnvh01 I have added FC17 also to Go library. Try it out and report problems in that repository. Thanks.