aldas/modbus-tcp-client #56: Illegal data address problem

pietrach, 2020-04-27:
I use Modbus master slave simulator

[image]

[image]

Code receive an error:

```
Packet to be sent (in hex): 8b7f00000006000400d60001
Binary received (in hex): 8b7f00000003008402
ModbusTcpClient\Exception\ModbusException Object ( [message:protected] => Illegal data address [string:Exception:private] => [code:protected] => 2 [file:protected] => D:\wamp\www\modbus\vendor\aldas\modbus-tcp-client\src\Packet\ResponseFactory.php [line:protected] => 85 [trace:Exception:private] => Array ( [0] => Array ( [file] => D:\wamp\www\modbus\test.php [line] => 28 [function] => parseResponseOrThrow [class] => ModbusTcpClient\Packet\ResponseFactory [type] => :: [args] => Array ( [0] => �� ) ) ) [previous:Exception:private] => [xdebug_message] => ( ! ) ModbusTcpClient\Exception\ModbusException: Illegal data address in D:\wamp\www\modbus\vendor\aldas\modbus-tcp-client\src\Packet\ResponseFactory.php on line 85 Call Stack #TimeMemoryFunctionLocation 10.0001401160{main}( )...\test.php:0 20.0392439400ModbusTcpClient\Packet\ResponseFactory::parseResponseOrThrow( )...\test.php:28 ) Illegal data address
#0 D:\wamp\www\modbus\test.php(28): ModbusTcpClient\Packet\ResponseFactory::parseResponseOrThrow('\x8B\x7F\x00\x00\x00\x03\x00\x84\x02') #1 {main}
```

aldas, 2020-04-27:
@pietrach is that test program set to emulate  'unit id'/'slave id' = 1 scenario?   If so then you need to create packet as
```
$unitId = 1;
$packet = new ReadInputRegistersRequest($startAddress, $quantity, $unitId);
```

pietrach, 2020-04-27:
@aldas result the same

```
Packet to be sent (in hex): 07b300000006010400d60001
Binary received (in hex): 07b300000003018402
ModbusTcpClient\Exception\ModbusException Object ( [message:protected] => Illegal data address [string:Exception:private] => [code:protected] => 2 [file:protected] => D:\wamp\www\modbus\vendor\aldas\modbus-tcp-client\src\Packet\ResponseFactory.php [line:protected] => 85 [trace:Exception:private] => Array ( [0] => Array ( [file] => D:\wamp\www\modbus\test.php [line] => 28 [function] => parseResponseOrThrow [class] => ModbusTcpClient\Packet\ResponseFactory [type] => :: [args] => Array ( [0] => �� ) ) ) [previous:Exception:private] => [xdebug_message] => ( ! ) ModbusTcpClient\Exception\ModbusException: Illegal data address in D:\wamp\www\modbus\vendor\aldas\modbus-tcp-client\src\Packet\ResponseFactory.php on line 85 Call Stack #TimeMemoryFunctionLocation 10.0002401160{main}( )...\test.php:0 20.0332439400ModbusTcpClient\Packet\ResponseFactory::parseResponseOrThrow( )...\test.php:28 ) Illegal data address
#0 D:\wamp\www\modbus\test.php(28): ModbusTcpClient\Packet\ResponseFactory::parseResponseOrThrow('\x07\xB3\x00\x00\x00\x03\x01\x84\x02') #1 {main}
```

[image]

pietrach, 2020-04-27:
It's working:

[image]

```
Packet to be sent (in hex): b41600000006010400d50001
Binary received (in hex): b41600000005010402003a
Parsed packet (in hex): b41600000005010402003a
Data parsed from packet (bytes):
Array ( [0] => 0 [1] => 58 ) Array ( [0] => 0 [1] => 58 ) Array ( [0] => 0 [1] => 58 ) 
```

aldas, 2020-04-27:
very good. I think it has to do how modbus slave program addressing is set

see:
[image]

pietrach, 2020-04-27:
But I don't understand:

[image]

Result is:
```

Packet to be sent (in hex): f35100000006010400d50001
Binary received (in hex): f351000000050104021104
Parsed packet (in hex): f351000000050104021104
Data parsed from packet (bytes):
Array ( [0] => 17 [1] => 4 )
Array ( [0] => 17 [1] => 4 )
Array ( [0] => 17 [1] => 4 ) 
```

Why ?

pietrach, 2020-04-27:
Ok, It's working

```
 $responseWithStartAddress = $response->withStartAddress($startAddress);
 print_r($responseWithStartAddress->getWordAt($startAddress)->getInt16());
```

return 4356

aldas, 2020-04-28:
Nice. You got everything working @pietrach? Can I close this ticket?

pietrach, 2020-04-28:
I think so but it isn't known what else will happen during :-)

closed 2020-05-22 by aldas