aldas/modbus-tcp-client #58: Call to a member function getWordAt() on null

marioshtika, 2020-06-03:
Hello,

I am getting this error message `Call to a member function getWordAt() on null`.

This is happening when I get a code 11 (Gateway targeted device failed to respond)

I am using the below code: 

```
$data = yield $socket->read(); // modbus packet is so small that one read is enough
if ($data === null) {
    return null;
}

return $request->parse($data);
```

It looks like when getting a "Gateway targeted device failed to respond", the `$data` variable it is not returning **null**,  nor a `ModbusException` or a `ModbusResponse`.

When I `var_dump($data)` it returns something like this `string(11) "j�\000\000\00"` (it is slightly different every time).

Is there a way to catch this error?

Thank you in advance.

aldas, 2020-06-03:
Are you using that amp example?

I assume you end up here https://github.com/aldas/modbus-tcp-client/blob/0930a98bb22ea65f3c6e9debdf56da70f485800d/src/Composer/Read/Register/ReadRegisterRequest.php#L68

```
    public function parse(string $binaryData)
    {
        $response = ResponseFactory::parseResponse($binaryData)->withStartAddress($this->request->getStartAddress());
        if ($response instanceof ErrorResponse) {
            return $response;
        }
```

parse will return ErrorResponse. In your code check if response is ErrorResponse and if it is then handle that case (discard/log/etc).

marioshtika, 2020-06-04:
Hello and thank you for the quick response.

Yes, I am using the amp example.

I already have that in my `try & catch`. But that is used after the `parse()` has run.

The problem is before parsing the data from `read`. It can not be parsed because (I am guessing) it is not returning the proper type.

This is happening only when I get a code 11 (Gateway targeted device failed to respond).

Is there something else I can do, to choose not to parse the data, if this error occurs.

marioshtika, 2020-06-04:
I followed the workflow on the ReadRequest.php file

```
/**
 * @param string $binaryData
 * @return array|ErrorResponse
 * @throws ModbusException
 * @throws \Exception
 */
public function parse(string $binaryData)
{
    $response = ResponseFactory::parseResponse($binaryData)->withStartAddress($this->request->getStartAddress());
    if ($response instanceof ErrorResponse) {
        return $response;
    }
    
    $result = [];
    foreach ($this->addresses as $address) {
        $result[$address->getName()] = $address->extract($response);
    }
    return $result;
}
```

and the response from this below line, it is not returning an `ErrorResponse`

```
$response = ResponseFactory::parseResponse($binaryData)->withStartAddress($this->request->getStartAddress());
```

This is the output of the `$respone` variable

```
class ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse#2197 (5) {
  private $data =>
  string(2) "\000�"
  private $dataBytes =>
  array(2) {
    [0] =>
    int(0)
    [1] =>
    int(236)
  }
  private $byteCount =>
  int(2)
  private $startAddress =>
  int(256)
  private $header =>
  class ModbusTcpClient\Packet\ModbusApplicationHeader#6133 (3) {
    private $transactionId =>
    int(42280)
    private $length =>
    int(5)
    private $unitId =>
    int(8)
  }
}
```

FYI, I am using the version **1.1.2**

closed 2020-07-27 by aldas