aldas/modbus-tcp-client #59: Display PLC data on a web page

Szabi76-hu, 2020-07-22:
As a beginner, I would like to ask what is required for writing and reading wago plc data on a web page in case of modbus tcp communication? I've seen some example projects here but don't know what files are needed for it? 
Thank you.
Szabolcs

aldas, 2020-07-22:
Hi, examples how to use this library can be found in this folder https://github.com/aldas/modbus-tcp-client/tree/master/examples

I do not know what you exactly mean by 'on a web page' but  simplest example to show modbus data on page would be
This is a example for situation here you would like to display humidity on a page. So humidity value is stored in PLC (ip: 192.168.0.1, port: 502) holding register with address 256 and its length is 1 register (16 bytes), with data type 16bit signed 
```
<?php

// Use composer dependency manager to get this library. See https://getcomposer.org/doc/00-intro.md
// I strongly suggest that you read what composer is. Please see
// https://www.youtube.com/watch?v=I6wm15OWyqg
// https://www.youtube.com/watch?v=d_zPOJ8cz_A
// https://www.codewall.co.uk/how-to-install-use-composer-tutorial-for-dummies-with-pdf/

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('192.168.0.1')
    ->build();

// our humidity value is stored in PLC holding register with address 256 and its length is 1 register (16 bytes), with data type 16bit signed integer
$startAddress = 256;
$quantity = 1;
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity);

try {
    $binaryData = $connection->connect()->sendAndReceive($packet);
    $responseWithStartAddress = ResponseFactory::parseResponseOrThrow($binaryData)->withStartAddress($startAddress);

    echo "<!DOCTYPE html><html><body><div>Humidity: " . ($responseWithStartAddress->getDoubleWordAt(256)->getInt16()) . "%<div></body></html>";

} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}
```

Szabi76-hu, 2020-07-22:
Yes, thank you for understanding the question. PLC -> local web server -> computer web browser: data. I will review the example I received and the composer program. Can I search you if I have a question?

aldas, 2020-07-22:
This is super fast way to init project

1. first download composer and install it https://getcomposer.org/download/
2. create folder for your project and go into it
```
mkdir test-project
cd test-project/
```
3. initialize composer file with composer
```
composer init
```
Output will be something similar
```
user@user:~/code/test-project$ composer init

                                            
  Welcome to the Composer config generator  
                                            

This command will guide you through creating your composer.json config.

Package name (<vendor>/<name>) [user/test-project]: 
Description []: 
Author [user <user@user.com>, n to skip]: n
Minimum Stability []: 
Package Type (e.g. library, project, metapackage, composer-plugin) []: project
License []: MIT

Define your dependencies.

Would you like to define your dependencies (require) interactively [yes]? no
Would you like to define your dev dependencies (require-dev) interactively [yes]? no

{
    "name": "user/test-project2",
    "type": "project",
    "license": "MIT",
    "require": {}
}

Do you confirm generation [yes]? yes
```
4. Add this modbus library to this new project with composer
```
composer require aldas/modbus-tcp-client
```
5. create folder for your own source code
```
mkdir src
```
6. create index.php file for our example code and copy code from previous comment into that file
```
touch src/index.php
# copy source code in previous comment to this index.php file
```
7. start php test http server in source code folder. This will server this project code in your machine port 8080
```
php -S localhost:8080 -t src/
```
8. Open your http://localhost:8080 in your favorite browser

aldas, 2020-07-25:
@Szabi76-hu did you get example running for yourself?

Szabi76-hu, 2020-07-27:
Yes, but unfortunately it doesn't work.

Szabi76-hu, 2020-07-27:
It says: 
 "An exception occurred address out of bounds #0 /var/www/index.php(30): ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse->getDoubleWordAt(165) #1 {main}"

aldas, 2020-07-27:
Is your quantity set to at least 2  i.e. `$quantity = 2;`?  `getDoubleWordAt` method needs at least 2 registers length of data. 

is this does not solve the problem then let me see the response from modbus.

add one 'echo between these lines.
```
    $binaryData = $connection->connect()->sendAndReceive($packet);
    echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;
    $responseWithStartAddress = ResponseFactory::parseResponseOrThrow($binaryData)->withStartAddress($startAddress);
```

aldas, 2020-07-27:
seems that it was my mistake. in previous example there was `$responseWithStartAddress->getDoubleWordAt(256)->getInt16()`

it should be `$responseWithStartAddress->getWordAt(256)->getInt16()` when `$quantity = 1;`

Szabi76-hu, 2020-07-27:
If I set its quantity to at least 2, it not work.

Inserting the echo line, type:

"Binary received (in hex): 6a8f000000050003020000 An exception occurred address out of bounds #0 /var/www/index.php(36): ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse->getDoubleWordAt(165) #1 {main}"

aldas, 2020-07-27:
is `$startAddress` same value  that is passed to getDoubleWordAt `getDoubleWordAt(165)` ?

try `$responseWithStartAddress->getWordAt($startAddress)->getInt16()`

Szabi76-hu, 2020-07-27:
If getWord:

Binary received (in hex): 054e000000050003020000
Humidity: 0%

Szabi76-hu, 2020-07-27:
The original code included:
"require __DIR__. '/../vendor/autoload.php';"
I had to change this:
"require __DIR__. '/vendor/autoload.php';"
because otherwise you did not load the page

aldas, 2020-07-27:
seems ok. atleast response from modbus has data bytes as `00 00` so  getting 0 seems ok.

Modbus packet contains following parts

``
    $binaryData = "\x05\x4e\x00\x00\x00\x05\x00\x03\x02\x00\x00";
``

* transaction id = `\x05\x4e`
* protocol (not used) = `\x00\x00`
* length of packet `\x00\x05`
* unit id = `\x00`
* function code = `\x03`   3 is read holding registers
* byte length of returned registers `\x02`
* registers data = `\x00\x00`

this file contains examples https://github.com/aldas/modbus-tcp-client/blob/de22a0763a1021746644aa90946fc4bffbd551b1/tests/unit/Packet/ResponseFactoryTest.php#L83

If you are really sure that address `165` should contain data then you can try requesting address `164` and `166`. Some PLC have a little different addressing logic. starting from 1 or starting from 0 and what not.

Szabi76-hu, 2020-07-27:
It works! Thank you very much! Are there any examle that write a value back to plc?

aldas, 2020-07-27:
see `examples/fc6.php`
https://github.com/aldas/modbus-tcp-client/blob/0930a98bb22ea65f3c6e9debdf56da70f485800d/examples/fc6.php#L17

Szabi76-hu, 2020-07-27:
Ok. Thank you very much!

closed 2020-07-27 by aldas