aldas/modbus-tcp-client #60: How to implement this from a server

vjdaya, 2020-08-04:
i m new to programming bro...  i m currently holding a website with cpanel... can i implement the modbus rtu over tcp by hosting the php files in my cpanel bro.. can u plz guide me... can u plz guide be if this is possible... i need to read my register values from some other location. can i use static IP for this process.. plz guide me... sorry if my idea sounds bizarre

aldas, 2020-08-07:
Hi @vjdaya , sorry for late reply. Assuming your cpanel server can connect to modbus port you can definitively use this library. Examples has RTU over TCP example - see https://github.com/aldas/modbus-tcp-client/blob/master/examples/rtu.php

vjdaya, 2020-08-07:
so do i need a static IP to communicate with my PLC. I m using a Select PLC and thanks to u i m currently able to communicate with my plc using your library with RTU over TCP example. thanks a lot. But can i communicate the same by assigning a static IP to my PLC RTU to TCP converter and read the values. Again sorry i m a new to programming. Thanks for your time.

aldas, 2020-08-07:
It depends on your IT infrastructure. Lets assume this scenario

1. Cpanel is hosted at some cloud/hosting provider server
2. PLC is hosted at your home/work (firewalled) network, behind some kind of router

Then PHP on cpanel server needs to know IP where PLC is located and there must be open port on that router and port forwarding setup from router to PLC internal IP.

If your some/work network does not have static IP you would use dynamic DNS provider to get fix DNS name for your changing IP.  This library can work even with DNS names `->setHost('mycoolplc.dyndnsprovider.com')`

first google search gives me this free DynDNS provider https://www.noip.com/free