aldas/modbus-tcp-client #65: Response Error

jplajpla23, 2020-10-16:
I run the example with React to Read one register from device and get an message error

`PHP Fatal error:  Uncaught ModbusTcpClient\Exception\ParseException: packet byte count does not match bytes in packet! count: 70, actual: 3 in /data/senslive3/vendor/aldas/modbus-tcp-client/src/Packet/ByteCountResponse.php:23
Stack trace:
#0 /data/senslive3/vendor/aldas/modbus-tcp-client/src/Packet/ModbusFunction/ReadHoldingRegistersResponse.php(32): ModbusTcpClient\Packet\ByteCountResponse->__construct('F\xF5\x00\x01', 0, 8185)
#1 /data/senslive3/vendor/aldas/modbus-tcp-client/src/Packet/ResponseFactory.php(48): ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse->__construct('F\xF5\x00\x01', 0, 8185)
#2 /data/senslive3/vendor/aldas/modbus-tcp-client/src/Composer/Read/ReadRequest.php(67): ModbusTcpClient\Packet\ResponseFactory::parseResponse('\x1F\xF9\x00\x00\x00\x06\x00\x03F\xF5\x00\x01')
#3 /data/senslive3/services/modbus/modbus_service.php(115): ModbusTcpClient\Composer\Read\ReadRequest->parse('\x1F\xF9\x00\x00\x00\x06\x00\x03F\xF5\x00\x01')
#4 /data/senslive3/vendor/evenement/evenement/src/Evenement/EventEm in /data/senslive3/vendor/aldas/modbus-tcp-client/src/Packet/ByteCountResponse.php on line 23

`

if I enbale my debug
I get this

`______________________________send_________________________
1ff900000006000346f50001
______________________________end send____________________

______________________________data receive_________________________
1ff900000006000346f50001
______________________________end data_________________________
`

aldas, 2020-10-16:
I think I have made typo with latest changes in example. 
Can you try change this line https://github.com/aldas/modbus-tcp-client/blob/aa72ff24da0b10f95edd84b269b54b72f390be1c/examples/example_parallel_requests_reactphp.php#L66

to
`$promise->resolve($request->parse($receivedData));`

if this does not help. try same address/quantity with https://github.com/aldas/modbus-tcp-client/blob/master/examples/fc3.php example

Although this may not help. Your response `1ff900000006000346f50001` seems strange. For quantity=1 response should be 1 byte shorter than request. 

Request structure is like:
```
 * Example packet: \x00\x01\x00\x00\x00\x06\x01\x03\x00\x6B\x00\x01
 * \x00\x01 - transaction id
 * \x00\x00 - protocol id
 * \x00\x06 - number of bytes in the message (PDU = ProtocolDataUnit) to follow
 * \x01 - unit id
 * \x03 - function code
 * \x00\x6B - start address
 * \x00\x01 - holding registers quantity to return
```
and Response like:
```
 * Example packet: \x81\x80\x00\x00\x00\x05\x01\x03\x02\xCD\x6B
 * \x81\x80 - transaction id
 * \x00\x00 - protocol id
 * \x00\x05 - number of bytes in the message (PDU = ProtocolDataUnit) to follow
 * \x01 - unit id
 * \x03 - function code
 * \x02 - returned registers byte count
 * \xCD\x6B - holding registers data (1 register)
```

aldas, 2020-10-16:
maybe logging response as would be more precise/granular:
```
                $connection->on('data', function ($data) use ($connection, $promise, $request, &$receivedData) {
                    $asHex = unpack('H*', $data)[1];
                    echo 'Binary received (in hex):   ' . $asHex . PHP_EOL;
```

jplajpla23, 2020-10-16:
Hi @aldas 
I update the plugin to last version
and try the example f03 and get

PHP Fatal error:  Uncaught Error: Call to undefined method ModbusTcpClient\Network\BinaryStreamConnection::sendPacket() in /data/test/services/modbus/test_mod.php:101
Stack trace:
#0 {main}
  thrown in /data/test/services/modbus/test_mod.php on line 101

if run with react the problem resists.

if I use the modbus pool slave softwre in windows I can read.
the real device can't.
but with modbus pool master i can read the real device.

aldas, 2020-10-16:
@jplajpla23 
ok, fc3 example was also using notexisting function `sendPacket`. It is fixed in main branch (no new version in packagist yet). Correct line is https://github.com/aldas/modbus-tcp-client/blob/29aafbc0e6031f4af739cff04af9481c17a3c108/examples/fc3.php#L23

I added more detailed logging to ReactPHP example. Please try with that example again.

I think fastest way could be to clone repo and just execute example
```
git clone https://github.com/aldas/modbus-tcp-client.git
cd modbus-tcp-client
composer install
composer require react/socket:^1.6
# before next command edit examples/example_parallel_requests_reactphp.php example with your ip/address etc and then execute example with
php examples/example_parallel_requests_reactphp.php
```

To help you diagnose problem I need to see what is exactly send and what is received (it maybe me multiple parts/chunks for one packet). This example will now echo this to output.

aldas, 2020-10-19:
@jplajpla23 are you having still problems?

jplajpla23, 2020-10-19:
I can't test today. may be only can test tomorrow or wednesday. I leave feddback after

jplajpla23, 2020-10-27:
Hi @aldas ,
 I run a sniffer(wireshark) and is the device return the same packet. I will debug the device.

closed 2020-10-27 by jplajpla23

aldas, 2020-10-27:
Hey @jplajpla23 I usually recommend when having problems with this library to test out sending requests to simulator ala https://www.modbusdriver.com/diagslave.html  just to single out/pinpoint problems - too see if is it device or is it library problem.