aldas/modbus-tcp-client #68: Receive connection from Modbus TCP

fernandomr, 2020-11-05:
Hi,
I've been running tests on this lib and connecting it to PyModSlave and it works perfectly.
But can this lib receive connections from a Modbus TCP device? There are devices that can send their data through Modbus TCP in certain periods of time.
Do you have an example on how to do that?
Thank you!

aldas, 2020-11-05:
so you would like to build Modbus server (respond to requests)?

This is new use case but I have this example https://github.com/aldas/modbus-tcp-client/blob/29aafbc0e6031f4af739cff04af9481c17a3c108/tests/integration/MockResponseServer.php#L35 that I use for intergration tests.  It opens port with ReactPHP and listens for requests and responds with mock responses.  

What you would need to do is to:

1. parse request
2. decide what functions it is, 
3. gather data 
4. create response instance 
5. respond with response

* look for this method how to parse https://github.com/aldas/modbus-tcp-client/blob/29aafbc0e6031f4af739cff04af9481c17a3c108/src/Packet/ResponseFactory.php#L27  this is for parsing responses but similar for request is not that hard to create
* look for classes to create requests ala FC3 https://github.com/aldas/modbus-tcp-client/blob/29aafbc0e6031f4af739cff04af9481c17a3c108/tests/unit/Packet/ModbusFunction/ReadHoldingRegistersRequestTest.php#L34

aldas, 2020-11-05:
See pull request #69 for example. I'll finish those classes for other function types in couple of days.

fernandomr, 2020-11-06:
Yes, a Modbus server (respond to Modbus TCP requests).
Thanks! I'll try it and I'll say if I find any trouble.

aldas, 2020-11-10:
@fernandomr I merged that branch into master. You can try example https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_response_server.php  Let me know if you have troubles. It might have typos or something that I forgot.

fernandomr, 2020-11-11:
@aldas thanks!
I'll try it! I couldn't test it before.

fernandomr, 2020-11-20:
Hi @aldas !
Your script has no issue, I must say. However, no tests I ran worked out (I've tested with qModMaster software), but because I got some aspects of Modbus wrong.
Excellent work on this repo, by the way.

I'll close this issue!
Thanks!

closed 2020-11-20 by fernandomr

aldas, 2020-11-20:
anyway, I created few days ago wiki page about debugging modbus. It is general enough to be used pretty most any library/program. At the end there is network traffic capture guide.  Maybe it helps https://github.com/aldas/modbus-tcp-client/wiki/Debugging-guide