aldas/modbus-tcp-client #70: Read total timeout expired

kimber4, 2020-11-12:
Hello

I'm getting this error. Anyone have an idea?

```
An exception occurred Read total timeout expired #0 /home/endsys.org/public_html/modbus/src/Network/BinaryStreamConnection.php(60): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom() #1 /home/endsys.org/public_html/modbus/src/Network/BinaryStreamConnection.php(81): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 /home/endsys.org/public_html/modbus/examples/1mb.php(41): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive() #3 {main}

This page loaded in 2 sec.
```

My codes:
```
$stime= microtime(TRUE);

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(xxx)
    ->setHost('x.x.x.x')
    ->setReadTimeoutSec(15) // increase read timeout to 3 seconds
    ->setWriteTimeoutSec(15) // increase read timeout to 3 seconds
    ->setConnectTimeoutSec(3)
    ->setTimeoutSec(3)
    ->build();

$startAddress = 1100;
$quantity = 4;
$unitID = 1;

    $packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $unitID);

        try {
            
            $binaryData = $connection->connect()->sendAndReceive($packet);

            $response = ResponseFactory::parseResponseOrThrow($binaryData);

            echo '<br />Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
            echo '<br />Data parsed from packet (bytes):' . PHP_EOL;
            print_r($response->getData());
            
            
            
        }
        catch(Exception $e) {
            $connection->close();
                echo '<br />READ ERROR' . PHP_EOL;
                echo '<br />An exception occurred' . PHP_EOL;
                echo $e->getMessage() . PHP_EOL;
                echo $e->getTraceAsString() . PHP_EOL;
        }
        finally
        {
            $connection->close();
        }

   echo "<br /><br />";

   $etime = microtime(TRUE);
   $time_t = ($etime- $stime);
   $time_t = round($time_t,0);
 
   echo 'This page loaded in '.$time_t.' sec.';
```

Timeout values 15 but page loaded in 2 sec. I think it could not find any stream_select. I'm trying more than 4 days.

aldas, 2020-11-12:
Please try debugging with different modbus server at first to better understand where the problem lies - is it client or server

For example:
1. Download modbus slave simulator from https://www.modbusdriver.com/diagslave.html `diagslave-3.2.zip` and unpack it somewhere. Start `win/diagslave.exe` from zip.
2. Set your host and port value in php example `>setHost('127.0.0.1')` and `->setPort(5022)`
3. Assuming that you are using values from previous step then modbus simulator should be started as

    ```bash
    diagslave -m tcp -p 5022
    ```

4. Run example against that simulator server and see if this gives different result.

aldas, 2020-11-12:
if you are using latest version (v2.1.0) please try  `2.0.1` and `master`

this will checkout version 2.0.1 for example:
```
git clone --branch 2.0.1 https://github.com/aldas/modbus-tcp-client.git modbus-tcp-client-201
cd modbus-tcp-client
composer install
# edit fc3.php example with your ip/address etc
php examples/fc3.php
```

kimber4, 2020-11-12:
> Please try debugging with different modbus server at first to better understand where the problem lies - is it client or server
> 
> For example:
> 
> 1. Download modbus slave simulator from https://www.modbusdriver.com/diagslave.html `diagslave-3.2.zip` and unpack it somewhere. Start `win/diagslave.exe` from zip.
> 2. Set your host and port value in php example `>setHost('127.0.0.1')` and `->setPort(5022)`
> 3. Assuming that you are using values from previous step then modbus simulator should be started as
>    ```shell
>    diagslave -m tcp -p 5022
>    ```
> 4. Run example against that simulator server and see if this gives different result.

I tried this simulator and my request worked fine. I edited StreamHandler.php and i saw, when stream_select worked foreach ($read as $stream) is not working.

Like this:

$read is not empty
Do stream_select
$read is empty

so, 

```
            if (!$dataReceived) {
                $timeSpentWaiting = microtime(true) - $lastAccess;
                if ($timeSpentWaiting >= $timeout) {
                    throw new IOException('Read total timeout expired');
                }
            }
```

working and giving error.

aldas, 2020-11-12:
have you checked if your machine is allowed actually to connect that ip:port?  it could be that your server/machine is firewalled. 

for example https://www.modbusdriver.com/modpoll.html can be used to test if another client can make that request
or here is different commands to see if you can connect to ip+port https://superuser.com/questions/621870/test-if-a-port-on-a-remote-system-is-reachable-without-telnet

kimber4, 2020-11-12:
Yes, I can connect with modbuspoll. I can share with you my ip address and port number if you want.

aldas, 2020-11-12:
use that logger from fc3.php and paste output here

```
require __DIR__ . '/logger.php';
...
...
...
    ->setLogger(new EchoLogger())
```

it outputs similar text as
```
2020-11-12 12:17:29.838783: [debug] msg: "Connected"
2020-11-12 12:17:29.934436: [debug] msg: "Data sent", data: array (
  1 => 'd41900000006000301000006',
)
2020-11-12 12:17:29.934616: [debug] msg: "Polling data"
2020-11-12 12:17:29.934716: [debug] msg: "Stream 31 @ index: 0 received data: ", data: array (
  1 => 'd4190000000f00030c000000000000000000000000',
)
```

aldas, 2020-11-12:
just to single out if it is this library networking problem you could try to use this 
https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_parallel_requests_reactphp.php  example. This uses ReactPHP networking code.

kimber4, 2020-11-12:
I think when i tried the ReactPHP codes, it's locked my device's port. I can't access now with modbuspoll or device's own TCP program.

kimber4, 2020-11-12:
$stream = @stream_socket_client("tcp://x.x.x.x:yyy", $errno, $errstr);
It's always working and I can do fwrite with your library's modbus packet but fread is alwas giving crash.

kimber4, 2020-11-12:
Just it was succeed like 3/50

aldas, 2020-11-12:
Seems Networking or OS problem to me.

If you have time and patience it would be nice to see what is actually send on ethernet level with packet inspection.

For windows https://www.wireshark.org/
On linux maybe using TCPDUMP `yum install tcpdump`/`apt install tcpdump` would be easier way.
See https://danielmiessler.com/study/tcpdump/ for examples and guide

sudo tcpdump -i enp7s0 -X host 192.168.1.100 and port 502

`enp7s0` is my ethernet interface
`192.168.1.100` is my modbus server (PLC)

start it one terminal ja do request on another.  and paste send and received TCP packets here

aldas, 2020-11-12:
just to be clear:
what is your:
* OS (version)
* PHP version
* this library version

aldas, 2020-11-12:
and if you are using
```
$connection = BinaryStreamConnection::getBuilder()
...
...
    ->setLogger(new EchoLogger())
    ->build();
```

please paste output here.  

It is big difference if there is no line similar to `2020-11-12 12:17:29.934716: [debug] msg: "Stream 31 @ index: 0 received data:`
or there is one.   If no lines are outputted then seems that `fread` will not received any bytes from network.   if there are similar lines then it could be that something is wrong with packed (data).

kimber4, 2020-11-12:
I can see send TCP packets but I can't see received TCP packets. Because I'm trying connect to a modem.

I'm trying same time on my local pc (Windows 10 Pro 64 bit (10.0, Build 19041) with xampp v3.2.4 (Apache/2.4.46 (Win64) OpenSSL/1.1.1g PHP/7.4.9)

and Ubuntu VPS Server with (LITESPEED/1.6.16 OPEN) and PHP 7.4

And I think library version is ## [2.0.1] - 2020-04-12 (from Changelog file last date)

kimber4, 2020-11-12:
I think my problem is this:,
a) If no lines are outputted then seems that fread will not received any bytes from network.

Because everything looking good to fwrite. When I came to fread it's returning empty and timed out. I read values succeed rarely.

kimber4, 2020-11-12:
```
$connection = BinaryStreamConnection::getBuilder()
...
...
    ->setLogger(new EchoLogger())
    ->build();
```

I tried this but nothing changed.

aldas, 2020-11-12:
strange. you could try running `diagslave -m tcp -p 5022` on a different machine (maybe on your local network) and see if php client works then. Too see if this problem is more tied to your modbus device or network overall.

aldas, 2020-11-12:
and packet inspection with WireShark or TCPDUMP would tell us if there is actually response from device or not.

kimber4, 2020-11-12:
Now device woke up and this is fc3 results:

`Packet to be sent (in hex): 9c44000000060003044c0004 2020-11-12 13:57:44.742968: [debug] msg: "Connected" 2020-11-12 13:57:44.743337: [debug] msg: "Data sent", data: array ( 1 => '9c44000000060003044c0004', ) 2020-11-12 13:57:45.744005: [debug] msg: "Polling data" An exception occurred Read total timeout expired #0 /home/xxx/public_html/modbus/src/Network/BinaryStreamConnection.php(60): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom() #1 /home/xxx/public_html/modbus/src/Network/BinaryStreamConnection.php(81): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 /home/xxx/public_html/modbus/examples/fc3new.php(24): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive() #3 {main}`

aldas, 2020-11-12:
your first example has `$unitID = 1;` but packet that is sent is ` \x9c\x44\x00\x00\x00\x06\x00\x03\x04\x4c\x00\x04`  byte after `\x06` is unit id (so for this packet it seems to be `0`).  

https://github.com/aldas/modbus-tcp-client/blob/5d554413342835fcf3562c2cdcff01fb46c92f25/src/Packet/ModbusFunction/ReadHoldingRegistersRequest.php#L16

Are you sure that you are addressing correct slave/unit and because of that response timeouts? Should it be `1` or `0`?

kimber4, 2020-11-12:
You are right. I forgot set the unitID, new report:
`Packet to be sent (in hex): aa70000000060003044c0004 2020-11-12 14:08:35.896906: [debug] msg: "Connected" 2020-11-12 14:08:35.897156: [debug] msg: "Data sent", data: array ( 1 => 'aa70000000060003044c0004', ) 2020-11-12 14:08:36.902421: [debug] msg: "Polling data" An exception occurred Read total timeout expired #0 /home/xxx/public_html/modbus/src/Network/BinaryStreamConnection.php(60): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom() #1 /home/xxx/public_html/modbus/src/Network/BinaryStreamConnection.php(81): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 /home/xxx/public_html/modbus/examples/fc3new.php(26): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive() #3 {main}`

aldas, 2020-11-12:
seems same - `aa70000000060003044c0004` is still `0` for unitID. and timeout seems to come very fast 
```
2020-11-12 14:08:35.897156 - connected
2020-11-12 14:08:36.902421 - timeout
```
its 1 sec.  increase that.

and if you are using cloned repo you could modify `src/Network/StreamHandler.php` code to move data debugging line of sooner
```
                if ($streamIndex !== null) {
                    $data = fread($stream, 256); // read max 256 bytes
                    if ($logger) {
                        $logger->debug("Stream {$streamId} @ index: {$streamIndex} received data: ", unpack('H*', $data));
                    }
                    if (!empty($data)) {

                        $packetData = ($result[$streamIndex] ?? b'') . $data;
```

just to see that is read. probably nothing.

I still suggest you to try packet inspection with Wireshark or TCPDUMP

kimber4, 2020-11-12:
Yes, I said at first message "Timeout values 15 but page loaded in 2 sec."

and I said

```
I tried this simulator and my request worked fine. I edited StreamHandler.php and i saw, when stream_select worked foreach ($read as $stream) is not working.

Like this:

$read is not empty
Do stream_select
$read is empty

so,

            if (!$dataReceived) {
                $timeSpentWaiting = microtime(true) - $lastAccess;
                if ($timeSpentWaiting >= $timeout) {
                    throw new IOException('Read total timeout expired');
                }
            }
working and giving error.
```

Probably after the steam_select, $read will be empty.

kimber4, 2020-11-12:
I sliced codes today and I reached this result.
$read clearing after the stream_select() so foreach ($read as $stream) is not working at StreamHandler.php

kimber4, 2020-11-12:
When I added eco lines at StreamHandler.php

```
            echo '<pre>'; print_r($read); echo '</pre>';
            
            $modifiedStreams = stream_select(
                $read,
                $write,
                $except,
                (int)$timeout,
                $timeoutUsec
            );
            if (false === $modifiedStreams) {
                throw new IOException('stream_select interrupted by an incoming signal');
            }

            if ($logger) {
                $logger->debug('Polling data');
            }

            echo '<pre>'; print_r($read); echo '</pre>';
```

```
2020-11-12 14:31:58.671488: [debug] msg: "Connected" 2020-11-12 14:31:58.672435: [debug] msg: "Data sent", data: array ( 1 => '797b00000006010301000006', )
Array
(
    [0] => Resource id #26
)
2020-11-12 14:31:58.980353: [debug] msg: "Polling data"
Array
(
)
An exception occurred Read total timeout expired #0 C:\xampp\htdocs\modbus-tcp-client-master\src\Network\BinaryStreamConnection.php(59): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom(Array, 0.3, Object(EchoLogger)) #1 C:\xampp\htdocs\modbus-tcp-client-master\src\Network\BinaryStreamConnection.php(80): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 C:\xampp\htdocs\modbus-tcp-client-master\examples\fc3.php(23): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive(Object(ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest)) #3 {main}
```

aldas, 2020-11-12:
`fread` being empty is totally ok. For example when PLC responses are slow then there could be multiple `fread` calls before data arrives or modbus packet is fragmented (arrives inside multiple TCP/IP packets)

what [stream_select](https://www.php.net/manual/en/function.stream-select.php) does is that is takes array of streams and waits their status to change - i.e. data to arrive, connection to closed etc. 
our timeout `Read total timeout expired` occurs then we did not receive any changes from `stream_select` or  received but none of them produced data from `fread`

this is basically same as what ReactPHP and Amp library does.

if [ReactPHP example](examples/example_parallel_requests_reactphp.php) does not work also for you then problem is  with PHP/OS/Device and you should inspect network traffic to determine if there are actually any bytes sent.

aldas, 2020-11-12:
In your last example `stream_select` returns with empty `$read` array. This probably means that `stream_select` did not receive any data - i.e modbus device is unable to respond due firewall.

```
Array
(
)
```

by adding `if ($modifiedStreams === 0)` check to after `stream_select` this timeout check is done sooner but result is same. 

```
            $modifiedStreams = stream_select(
                $read,
                $write,
                $except,
                (int)$timeout,
                $timeoutUsec
            );
            if ($modifiedStreams === 0) {
                throw new IOException('Read total timeout expired without receiving anything from streams');
            }
            if (false === $modifiedStreams) {
                throw new IOException('stream_select interrupted by an incoming signal');
            }
```

You should really check if on network level modbus device actually sends bytes as response.

kimber4, 2020-11-12:
`36	6.039452	192.168.16.127	178.242.106.116	TCP	66	64098 → 502 [SYN] Seq=0 Win=64240 Len=0 MSS=1460 WS=256 SACK_PERM=1`

https://www.transfernow.net/OWKm0q112020

You can search 178.242.106.116 and PHP output:
```
2020-11-12 15:07:15.285520: [debug] msg: "Connected" 2020-11-12 15:07:15.295109: [debug] msg: "Data sent", data: array ( 1 => '643e00000006010301000006', )
Array
(
    [0] => Resource id #26
)
2020-11-12 15:07:15.609020: [debug] msg: "Polling data"
Array
(
)
An exception occurred Read total timeout expired #0 C:\xampp\htdocs\modbus-tcp-client-master\src\Network\BinaryStreamConnection.php(59): ModbusTcpClient\Network\BinaryStreamConnection->receiveFrom(Array, 0.3, Object(EchoLogger)) #1 C:\xampp\htdocs\modbus-tcp-client-master\src\Network\BinaryStreamConnection.php(80): ModbusTcpClient\Network\BinaryStreamConnection->receive() #2 C:\xampp\htdocs\modbus-tcp-client-master\examples\fc3.php(23): ModbusTcpClient\Network\BinaryStreamConnection->sendAndReceive(Object(ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest)) #3 {main}
```

aldas, 2020-11-12:
[image]

looking at packets it will seems that  client closes connection after 0.3seconds.  `0.3` is acutally default value for `$readTimeoutSec`
https://github.com/aldas/modbus-tcp-client/blob/5d554413342835fcf3562c2cdcff01fb46c92f25/src/Network/BinaryStreamConnectionProperties.php#L31

also establishing connection is quite slow `0.6s` for this case. Default timeout value for connection is `$connectTimeoutSec = 1;`

-----------
As this modbus server seems to be connected to internet (VERY BAD PRACTICE!!!) I tried to connect and send packet myself

with php library and modpoll - both timeout when waiting for answer.

```
./modpoll -m tcp -a 1 -r 256 -c 6 -t 3 -1 -o 10 xxx.xxx.xxx.xxx
modpoll 3.6 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2018 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP
Slave configuration...: address = 1, start reference = 256, count = 6
Communication.........: xxx.xxx.xxx.xxx, port 502, t/o 10.00 s, poll rate 1000 ms
Data type.............: 16-bit register, input register table

-- Polling slave...
Reply time-out!
```

`sudo tcpdump -i enp7s0 -X host xxx.xxx.xxx.xxx and port 502`

```
20:58:23.437849 IP XXXX.lan.49200 > 178.242.106.116.502: Flags [S], seq 210715453, win 64240, options [mss 1460,sackOK,TS val 1300591401 ecr 0,nop,wscale 9], length 0
        0x0000:  4500 003c de80 4000 4006 7c98 c0a8 0194  E..<..@.@.|.....
        0x0010:  XXXX XXXX c030 01f6 0c8f 433d 0000 0000  ..jt.0....C=....
        0x0020:  a002 faf0 dfd1 0000 0204 05b4 0402 080a  ................
        0x0030:  4d85 7329 0000 0000 0103 0309            M.s)........
20:58:23.805117 IP 178.242.106.116.502 > XXXX.lan.49200: Flags [S.], seq 1250, ack 210715454, win 10880, options [mss 1360,nop,wscale 0,nop,nop,sackOK], length 0
        0x0000:  4500 0034 83cd 4000 7306 a453 XXXX XXXX  E..4..@.s..S..jt
        0x0010:  c0a8 0194 01f6 c030 0000 04e2 0c8f 433e  .......0......C>
        0x0020:  8012 2a80 4e73 0000 0204 0550 0103 0300  ..*.Ns.....P....
        0x0030:  0101 0402                                ....
20:58:23.805192 IP XXXX.lan.49200 > 178.242.106.116.502: Flags [.], ack 1, win 126, length 0
        0x0000:  4500 0028 de81 4000 4006 7cab c0a8 0194  E..(..@.@.|.....
        0x0010:  XXXX XXXX c030 01f6 0c8f 433e 0000 04e3  ..jt.0....C>....
        0x0020:  5010 007e dfbd 0000                      P..~....
20:58:23.805343 IP XXXX.lan.49200 > 178.242.106.116.502: Flags [P.], seq 1:13, ack 1, win 126, length 12
        0x0000:  4500 0034 de82 4000 4006 7c9e c0a8 0194  E..4..@.@.|.....
        0x0010:  XXXX XXXX c030 01f6 0c8f 433e 0000 04e3  ..jt.0....C>....
        0x0020:  5018 007e dfc9 0000 0001 0000 0006 0104  P..~............
        0x0030:  00ff 0006                                ....
20:58:24.100011 IP 178.242.106.116.502 > XXXX.lan.49200: Flags [.], ack 13, win 10868, length 0
        0x0000:  4500 0028 83ce 4000 7306 a45e XXXX XXXX  E..(..@.s..^..jt
        0x0010:  c0a8 0194 01f6 c030 0000 04e3 0c8f 434a  .......0......CJ
        0x0020:  5010 2a74 8eda 0000 0000 0000 0000       P.*t..........
20:58:33.812649 IP XXXX.lan.49200 > 178.242.106.116.502: Flags [F.], seq 13, ack 1, win 126, length 0
        0x0000:  4500 0028 de83 4000 4006 7ca9 c0a8 0194  E..(..@.@.|.....
        0x0010:  XXXX XXXX c030 01f6 0c8f 434a 0000 04e3  ..jt.0....CJ....
        0x0020:  5011 007e dfbd 0000                      P..~....
20:58:34.581246 IP 178.242.106.116.502 > XXXX.lan.49200: Flags [.], ack 14, win 10880, length 0
        0x0000:  4500 0028 83cf 4000 7306 a45d XXXX XXXX  E..(..@.s..]..jt
        0x0010:  c0a8 0194 01f6 c030 0000 04e3 0c8f 434b  .......0......CK
        0x0020:  5010 2a80 8ecd 0000 0000 0000 0000       P.*...........
20:58:34.603632 IP 178.242.106.116.502 > XXXX.lan.49200: Flags [F.], seq 1, ack 14, win 10880, length 0
        0x0000:  4500 0028 83d0 4000 7306 a45c XXXX XXXX  E..(..@.s..\..jt
        0x0010:  c0a8 0194 01f6 c030 0000 04e3 0c8f 434b  .......0......CK
        0x0020:  5011 2a80 8ecc 0000 0000 0000 0000       P.*...........
20:58:34.603684 IP XXXX.lan.49200 > 178.242.106.116.502: Flags [.], ack 2, win 126, length 0
        0x0000:  4500 0028 0000 4000 4006 5b2d c0a8 0194  E..(..@.@.[-....
        0x0010:  XXXX XXXX c030 01f6 0c8f 434b 0000 04e4  ..jt.0....CK....
        0x0020:  5010 007e b8ce 0000                      P..~....
^C
9 packets captured
9 packets received by filter
0 packets dropped by kernel
```

Either it is network connection or that modbus device is not working properly

aldas, 2020-11-12:
Hope you do not ming. I deleted your comment with modbus public IP.

kimber4, 2020-11-13:
I have a more device with static IP address. This device sometimes blocking connection when we tried too much. Can you try another IP address? Because I can connect with modbuspoll If device is working normal.

Modbuspoll can connect and collect data but why our library is not working. What is the difference, I don't know.

kimber4, 2020-11-13:
I found the error. We need sleep after the connection. It's work fine now. Thank you so much for your helps. May you delete my first edited message because its including my domain name and please delete in your text like

`20:58:34.603684 IP XXXX.lan.49200 > x.x.x.x: `

aldas, 2020-11-13:
@kimber4 I can not delete first one, can only edit it but edit history is still there.  I'll delete whole issue.