aldas/modbus-tcp-client #71: RTU over TCP: Read total timeout expired

huyvpc, 2020-12-08:
Dear Sir,

I run diagslave.exe as slave simulator, then I test by Simply TCP Client to write and read holding register and it's work
[image]

[image]

But get error when run php. 
`use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadHoldingRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;
use ModbusTcpClient\Packet\RtuConverter;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->setReadTimeoutSec(5) // increase read timeout to 3 seconds
    ->build();

$startAddress = 0;
$quantity =10;
$slaveId = 1; // RTU packet slave id equivalent is Modbus TCP unitId

$tcpPacket = new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveId);
$rtuPacket = RtuConverter::toRtu($tcpPacket);

$connection->connect()->send($rtuPacket);
usleep(300000); // sleep 0.3 seconds
$binaryData = $connection->receive();
    
echo 'RTU Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

$response = RtuConverter::fromRtu($binaryData);
echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
echo 'Data parsed from packet (bytes):' . PHP_EOL;
print_r($response->getData());

$connection->close();`

The error show below:
[image]

I check in slave simulator when read form tcp client and php show the same comment as the below picture

[image]

Could you give me some advise, thanks!

Huy Bui

aldas, 2020-12-08:
I think problems is that by default receive waits for full TCP packet but RTU packet is different.

Try creating connection object as this:
```php
$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->setReadTimeoutSec(3) 
    ->setIsCompleteCallback(function ($binaryData, $streamIndex) {
        return true; // do not check for complete TCP packet structure
    })
    ->build();
```

huyvpc, 2020-12-08:
It's fantastic, thanks Sir for quick response!

Huy Bui

huyvpc, 2020-12-16:
Dear aldas,

I tried connect to power metter via moxa converter and the error come back.
See below pictures as config.

[image]
[image]

Please help!

Thanks!

aldas, 2020-12-16:
maybe raising read timeout helps. from the picture I see it is 0.34s converter but library default is 0.3s

```
->setReadTimeoutSec(0.3) // timeout when waiting response from server
```

From these pictures it is hard to tells what could be wrong. 

You could use debugging guide and capture network traffic (https://github.com/aldas/modbus-tcp-client/wiki/Debugging-guide#capture-network-traffic-to-see-what-is-actually-sendreceived)  In case PHP is disconnecting before everything is received this would help and increasing wait times could give valuable information.

huyvpc, 2020-12-16:
I changed timeout but no luck, I tried run without rtu request, it returns result that doesn't seem correct

$ip = '192.168.1.100';
$port = 502;
$connection = BinaryStreamConnection::getBuilder()
    ->setPort($port)
    ->setHost($ip)
    ->setReadTimeoutSec(1) // increase read timeout to 3 seconds
    ->setIsCompleteCallback(function ($binaryData, $streamIndex) {
        return true; // do not check for complete TCP packet structure
    })
    ->build();
    
$startAddress = 0;
$quantity =10;
$slaveId = 1; // RTU packet slave id equivalent is Modbus TCP unitId

$tcpPacket = new ReadHoldingRegistersRequest($startAddress, $quantity, $slaveId);

$connection->connect()->send($tcpPacket);
usleep(300000); // sleep 0.3 seconds
$binaryData = $connection->receive();
    
echo 'RTU Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

$response = ResponseFactory::parseResponse($binaryData);
//echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
echo 'Data parsed from packet (bytes):' . PHP_EOL;
print_r($response->getData());
$connection->close();

The result is bellow:
[image]

I test with Simply Modbus Client then result looks good

[image]

How can I get same result as Client app?

Thanks!

huyvpc, 2020-12-16:
This is the traffic monitoring of Moxa converter:

[image]

Thanks!

aldas, 2020-12-16:
So when using TCP you received response that could be understood. Its seems that our converter is using Modbus TCP instead of RTU. And from data I see that addressing (startaddress) is a little different (+1/-1).   From Simple Modbus picture it seems that you are asking start address 0(+1 offset) and that is why response is different from PHP response.   Add +1 in your PHP code to start address and result should match.

aldas, 2020-12-16:
Modbus packet structure is
```
 * Example packet: \x00\x01\x00\x00\x00\x06\x01\x03\x00\x6B\x00\x01
 * \x00\x01 - transaction id
 * \x00\x00 - protocol id
 * \x00\x06 - number of bytes in the message (PDU = ProtocolDataUnit) to follow
 * \x01 - unit id
 * \x03 - function code
 * \x00\x6B - start address
 * \x00\x01 - holding registers quantity to return
```

`05 5a 00 00 00 06 01 03 00 00 00 0a` - php request 
`05 5a 00 00 00 06 01 03 ff ff 00 0a` - Simply modbus request 

Seems that Simply modbus does  start address = uint16(0 - 1) => 0xFF 0xFF

This start address being 1 number different is due that some modbus implementations start addressing from 1 instead of 0.  I do not know that is right for your case.

huyvpc, 2020-12-16:
If you can see, the number of element of results in php is doubled (20).

aldas, 2020-12-16:
This is because you print bytes `print_r($response->getData());`  1 register is 2 bytes. `getData` returns response data as array of bytes.

If you want register/words converted to some type then use:
```
    foreach ($response as $word) {
        print_r($word->getInt16()); // assuming you want to convert register bytes to int16
    }
```

or 
```
$registers = $response->getWords();
```

See https://github.com/aldas/modbus-tcp-client/blob/master/src/Packet/Word.php#L22 that types exists for Word.

You may even use https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_parallel_requests.php to parse response to named array of converted types.

aldas, 2020-12-16:
And see https://github.com/aldas/modbus-tcp-client/blob/master/examples/fc3.php#L34 lines where different methods are used to access register data

huyvpc, 2020-12-17:
Thanks aldas

I read holding register from address of 30000 as device datasheet but it raise error

[image]

[image]

Here is deovice datasheet.

[OP - EM2M-1P-C-100A.pdf](https://github.com/aldas/modbus-tcp-client/files/5706753/OP.-.EM2M-1P-C-100A.pdf)

Do you have any glue?

huyvpc, 2020-12-17:
Did I miss some config in converter or device so it raised error when send rtu package to read register?

huyvpc, 2020-12-18:
Dear aldas,

I tried with ReadInputRegistersRequest from address of 30000, it return binary string, could I convert it to readable value?

Thanks!

huyvpc, 2020-12-18:
Dear Sir,

I use Simply and  changed minus offset then it's work. Can I change it in PHP?

aldas, 2020-12-18:
from your picture it seems that you have `first register = 30000` and `minus offset = 30000`  which means that you are actually asking for first register = 0. It may display 30000, 30001 ... in that result table column `register #` but is is actually 0,1,...  as 30000-30000=0

My suggestion - set `minus offset = 0` and do not use it.   When you are not using register offset then `first register` value is same as PHPs `$startAddress`

aldas, 2020-12-18:
I suggest you to read from **How is data stored in Standard Modbus?** block at https://www.simplymodbus.ca/FAQ.htm#Stored

> For example, the first Holding Register, number 40001, has the Data Address 0000.
The difference between these two values is the offset.
Each table has a different offset. 1, 10001, 30001 and 40001.

PHP is dealing with "data address" that is "Register number - Register offset".

huyvpc, 2020-12-18:
You're right Sir,

30xxx for input registrers and 40xxx for holding registers. But I wonder why they use 2 offset for 1 value or that merge to int32?

[image]

Thanks!

aldas, 2020-12-18:
I assume that these values are stored as Int32 values (2 register/word = 4 bytes). 
NB: if you need to get int32 value 

```php
$address = 2;
...
...
$response = ResponseFactory::parseResponseOrThrow($binaryData)->withStartAddress($startAddress);
$doubleWord = $response->getDoubleWordAt($address);
$int32 = $doubleWord->getInt32();
```
or with [builder](https://github.com/aldas/modbus-tcp-client/blob/master/examples/example_parallel_requests.php)

```php
$address = 2;
$fc3 = ReadRegistersBuilder::newReadHoldingRegisters('tcp://127.0.0.1:5022')
    ->int32($address, 'voltage_phase_2')
    ->build(); 
```

huyvpc, 2020-12-22:
Dear Sir,

It's stored in float and now It works like charm!

Thank you so much for all your help Sir!

closed 2020-12-24 by aldas