aldas/modbus-tcp-client #73: sendAndReceive works endlessly (timeout is ignored)

galpeaceman, 2021-01-25:
I have a working app using this module.
It works perfectly for days, sampling thousands of samples from a single device.
Then, one day, the CPU goes up to 100% and works endlessly until I kill it.

I think I understand the source of the problem.
See ModbusTcpClient\Network\StreamHandler's receiveFrom method (I copied here the relevant part of the code):

```
        if ($timeout === null) {
            $timeout = 0.3;
        }
        ...
        $lastAccess = microtime(true);
        ...
        while ($responsesToWait > 0) {
            ...
            $dataReceived = false;
            foreach ($read as $stream) {
                ...
                if ($index !== null) {
                    ...
                    $data = fread($stream, 256); // read max 256 bytes
                    if (!empty($data)) {
                        ...
                        $dataReceived = true;
                    }
                }
            }
            if (!$dataReceived) {
                $timeSpentWaiting = microtime(true) - $lastAccess;
                if ($timeSpentWaiting >= $timeout) {
                    throw new IOException('Read total timeout expired');
                }
            }
            $lastAccess = microtime(true);
```

So, what if fread fails very quickly (due to a connection timeout) and $dataReceived is false;
$timeSpentWaiting would be a very small number, possibly less than $timeout.
No matter, after a few tries $timeout would pass, right? No. It will never pass. Why?
Because $lastAccess is reset again to microtime(true) and the whole thing repeats itself.

I think "$lastAccess = microtime(true);" should be moved inside "if (!$dataReceived) {", fater the timeSpent test completes.

What do you say?

aldas, 2021-01-25:
first idea would be - maybe machine clock is being synchronized "back in time". I have to investigate how php acts when clock 'drifts'.     Do you have NTP or some automatic time synchronization setup for your server?

galpeaceman, 2021-01-25:
Though 'drifting' is possible, it is not required to explain the phenomena; since the $lastAccess resets itself at every iteration, it is very possible that $timeout did not pass yet

aldas, 2021-01-25:
You are correct. It is definitely problem when `stream_select` returns early and `fread` does not produce data.  Probably something happens with connection. I guess `fread` would return boolean `false` in that case for you.

Can you modify your code (vendor folder) in production? And does that problem occur reliably?

You could try
```php
                    $data = fread($stream, 256); // read max 256 bytes
                    if ($data === false) {
                        throw new IOException('fread error during receiveFrom');
                    }
                    if (!empty($data)) {
```

aldas, 2021-01-25:
It is in master branch now. I'll do proper release in couple of days

galpeaceman, 2021-01-25:
I have made the change and lets wait and see.
Unfortunately, this does not reproduce often; last time it happened was 6 weeks ago.
Let's assume this will solve the problem, and if it persists, I will reopen the issue.
Thanks!

aldas, 2021-01-25:
@galpeaceman are you really running same PHP process for weeks? I use this library in industrial environment (4+ year now) and out of fears of memory leaks and stuff I built my modbus poller to run X amount of time and then automatically stop/restart + guards not to run multiple instance etc. I think I'm quite paranoid :)

galpeaceman, 2021-01-25:
Well, I did not mean it is the same PHP process.. it currently resets after 5 reads, just like you described.
But we do intend to increase the read count per process greatly, once we solve the crash, we will do that.
I just meant that it makes a read every minute for a few weeks without errors, and then one morning it just freezes.

aldas, 2021-01-29:
New version is published - I'll close this issue for now.  And @galpeaceman thanks for pointing it out!

closed 2021-01-29 by aldas