aldas/modbus-tcp-client #76: Problem  connecting F03

dcorrea-estrav, 2021-03-03:
Hello @aldas ,

I have a problem connecting F03 I don't get any data when trying to connect, can you help me?

* Environment:
   - OS: Ubuntu 18.04
   - PHP version 7.4.10
   - Library version 2.1.1

my code: 

```
  $connection = BinaryStreamConnection::getBuilder()
      ->setPort(1346)
      ->setHost('192.168.1.24')
      ->setConnectTimeoutSec(1.5) // timeout when establishing connection to the server
      ->setWriteTimeoutSec(0.5) // timeout when writing/sending packet to the server
      ->setReadTimeoutSec(3) // timeout when waiting response from server
      ->setIsCompleteCallback(function ($binaryData, $streamIndex) {
          return true; // do not check for complete TCP packet structure
      })
      ->build();

  $startAddress = 1003;
  $quantity = 2;
  $unitID = 2;
  $packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $unitID);

  try {
      $binaryData = $connection->connect()->sendAndReceive($packet);
      /**
       * @var $response ReadHoldingRegistersResponse
       */
      $response = ResponseFactory::parseResponseOrThrow($binaryData);

      print_r($response->getData());
      foreach ($response as $word) {
          print_r($word->getBytes());
      }
      foreach ($response->asDoubleWords() as $doubleWord) {
          print_r($doubleWord->getBytes());
      }

      // set internal index to match start address to simplify array access
      $responseWithStartAddress = $response->withStartAddress($startAddress);
      print_r($responseWithStartAddress[$startAddress]->getBytes()); // use array access to get word
      print_r($responseWithStartAddress->getDoubleWordAt($startAddress)->getFloat());

  } catch (Exception $exception) {
      echo 'An exception occurred' . PHP_EOL;
      echo $exception->getMessage() . PHP_EOL;
      echo $exception->getTraceAsString() . PHP_EOL;
  } finally {
      $connection->close();
  }
```

response:
`An exception occurred Read total timeout expired #0`

device configuration:
[image]
 

what am I doing wrong ?

aldas, 2021-03-03:
as a first thing I would suggest to use logger that outputs more connection related info
This will echo more stuff.
```php
->setLogger(new EchoLogger())
```
I would need to see that logger output to help.

as a next thing I would suggest to verify if that IP+port is actually listening.  You can combine this step with trying other clients too see if it is PHP related or not.

I have written debugging guide and there is block to use another client to get "second" opinion. See this https://github.com/aldas/modbus-tcp-client/wiki/Debugging-guide#2-try-another-modbus-client

Third:
Is that program on image [USR-TCP232-xxx](https://www.pusr.com/download/M4/USR-TCP232-M4-EN%20V1.0.6.pdf) . Maybe you are reading/interpreting port number wrong?

dcorrea-estrav, 2021-03-03:
Hello @aldas thanks for your answer,

**My code:**

[image]

**Response:**

```
2021-03-03 19:14:56.032294: [debug] msg: "Connected" 
2021-03-03 19:14:56.033539: [debug] msg: "Data sent", data: array ( 1 => '45e000000006020303eb0002', ) 
2021-03-03 19:14:59.035849: [debug] msg: "Polling data" An exception occurred Read total timeout expired #0
```

Nmap
[image]

aldas, 2021-03-03:
mmm, I would definitely now suggest trying other client - just to see if this problem is limited to PHP or affects other clients. 

If you choose to use https://github.com/aldas/modbus-tcp-client/wiki/Debugging-guide#2-try-another-modbus-client

```bash
./modpoll -m tcp -a 2 -r 1003 -c 2 -t 4 -1 -o 10 192.168.1.24
```

dcorrea-estrav, 2021-03-03:
I have tried with a third party application and it gets the data correctly, as shown in the image

[image]

sorry for so many questions and thanks for your answers

aldas, 2021-03-03:
Are you sure that proxy server (modbus bride) is not doing modbus RTU over TCP?  I would recommend triring `modpoll` and that `-tcp` flag says that it will use   modbus tcp.  `-m rtu ` would use mobus RTU.

aldas, 2021-03-03:
if it turns out to be RTU then you should use https://github.com/aldas/modbus-tcp-client/blob/master/examples/rtu.php as implementation

dcorrea-estrav, 2021-03-03:
As you indicated, the device is in Modbus TCP mode. Now I get the following information: 

```
2021-03-03 19:45:25.544735: [debug] msg: "Connected" 
2021-03-03 19:45:25.545476: [debug] msg: "Data sent", data: array ( 1 => '661d00000006020303eb0002', ) 
2021-03-03 19:45:25.578589: [debug] msg: "Polling data" 
2021-03-03 19:45:25.578663: [debug] msg: "Stream 4 @ index: 0 received data: ", data: array ( 1 => '661d00000003028302', )
```

I should get a value similar to 339.0 0 or 399, what data from the received should I convert and how to do it?

aldas, 2021-03-03:
I assume it is stored as float  (2 registers,  32bit value)?

```php
    $responseWithStartAddress = $response->withStartAddress($startAddress);
    print_r($responseWithStartAddress->getDoubleWordAt($startAddress)->getFloat());
```

if is is  `int32` or `uint32` then use `getInt32()` etc methods and divide by 10 or 100 or whatever the scale is.

aldas, 2021-03-03:
looking at that response `661d00000003028302` I think it is a error response. https://github.com/aldas/modbus-tcp-client/blob/0ce52f3a7ade81f34c470249ea7dc611b5f4359f/src/Packet/ErrorResponse.php#L12

error code `02` is 'Illegal data address';

dcorrea-estrav, 2021-03-03:
Hello @aldas,

I have already made it work as it should, this could not have been achieved without people like you, thank you very much for your help.
my sincere thanks, you've got me out of a bind. Blessings

**my response code:**
```
2021-03-03 20:20:23.781698: [debug] msg: "Connected" 
2021-03-03 20:20:23.782943: [debug] msg: "Data sent", data: array ( 1 => '52e600000006020403eb0002', ) 
2021-03-03 20:20:23.819809: [debug] msg: "Polling data" 
2021-03-03 20:20:23.819888: [debug] msg: "Stream 5 @ index: 0 received data: ", data: array ( 1 => '52e600000007020404129e0000', ) Array ( [0] => 18 [1] => 158 [2] => 0 [3] => 0 ) Array ( [0] => 18 [1] => 158 ) Array ( [0] => 0 [1] => 0 ) 4766Array ( [0] => 18 [1] => 158 ) 4766 
```

aldas, 2021-03-03:
seems to be (u)int32 value

```php
    public function testShouldParseReadHoldingRegistersResponse()
    {
        //trans + proto + len   + uid + fc + qnt + data
        //52 e6 + 00 00 + 00 07 + 02  + 04 + 04  + 12 9e 00 00
        $data = "\x52\xe6\x00\x00\x00\x07\x02\x04\x04\x12\x9e\x00\x00";

        /** @var ReadHoldingRegistersResponse $response */
        $response = ResponseFactory::parseResponse($data);

        $this->assertEquals(4766, $response->getDoubleWordAt(0)->getInt32());
    }
```

closed 2021-03-04 by aldas