aldas/modbus-tcp-client #79: FC3 cannot read single register value

ndcisiv, 2021-04-05:
I can't seem to figure out how to read a single register value (1001) using the examples.  Here is my code

`$connection = BinaryStreamConnection::getBuilder()
            ->setPort(502)
            ->setHost('10.122.144.100')
            ->setConnectTimeoutSec(1.5) // timeout when establishing connection to the server
            ->setWriteTimeoutSec(0.5) // timeout when writing/sending packet to the server
            ->setReadTimeoutSec(0.3) // timeout when waiting response from server
            ->setLogger(new \EchoLogger())
            ->build();

        $startAddress = 1001;
        $quantity = 4;
        $unitID = 0;
        $packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $unitID);

        try {
            $binaryData = $connection->connect()->sendAndReceive($packet);

            /**
             * @var $response ReadHoldingRegistersResponse
             */
            $response = ResponseFactory::parseResponseOrThrow($binaryData);

            print_r($response->getData());
            foreach ($response as $word) {
                print_r($word->getBytes());
            }
            foreach ($response->asDoubleWords() as $doubleWord) {
                print_r($doubleWord->getBytes());
            }

            // set internal index to match start address to simplify array access
            $responseWithStartAddress = $response->withStartAddress($startAddress);
            print_r($responseWithStartAddress[$startAddress]->getBytes()); // use array access to get word
            print_r($responseWithStartAddress->getDoubleWordAt($startAddress)->getFloat());

        } catch (Exception $exception) {
            echo 'An exception occurred' . PHP_EOL.$exception->getMessage() . PHP_EOL.$exception->getTraceAsString() . PHP_EOL;
        } finally {
            $connection->close();
        }`

I only want the single register value and tried using `$quantity = 1;` but it gave me an error saying:

"getDoubleWords needs packet byte count to be multiple of 4"

When I set the quantity to 4 I get this response:

2021-04-05 15:58:25.312455: [debug] msg: "Connected"
2021-04-05 15:58:25.312661: [debug] msg: "Data sent", data: array (
  1 => '1f7c00000006000303e90004',
)
2021-04-05 15:58:25.354524: [debug] msg: "Polling data"
2021-04-05 15:58:25.354613: [debug] msg: "Stream 282 @ index: 0 received data: ", data: array (
  1 => '1f7c0000000b0003080000000000000000',
)
Array
(
    [0] => 0
    [1] => 0
    [2] => 0
    [3] => 0
    [4] => 0
    [5] => 0
    [6] => 0
    [7] => 0
)
Array
(
    [0] => 0
    [1] => 0
)
Array
(
    [0] => 0
    [1] => 0
)
Array
(
    [0] => 0
    [1] => 0
)
Array
(
    [0] => 0
    [1] => 0
)
Array
(
    [0] => 0
    [1] => 0
    [2] => 0
    [3] => 0
)
Array
(
    [0] => 0
    [1] => 0
    [2] => 0
    [3] => 0
)
Array
(
    [0] => 0
    [1] => 0
)
0

I verified using the modpoll command `modpoll -a 1 -c 1 -t 4 -r 1001 -m tcp 10.122.144.100` which gives me the output:

modpoll 3.9 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2020 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 0, start reference = 1001, count = 1
Communication.........: 10.122.144.100, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[1001]: 90

The response should be a 16bit unsigned integer from the device documentation.

What am I missing here?

ndcisiv, 2021-04-05:
I figured it out.  The device register value I am trying to access is 1001, however using this script I need to ask for 1000 in order to get the correct value.  It is working perfectly now!  Thank you for this awesome tool.

closed 2021-04-05 by ndcisiv

aldas, 2021-04-05:
ok, nice to hear that you got it working.

`1001` vs `1000`  - modbus addresses are 0-based in modbus protocol. In documentation they are sometimes/often referred as 1-based. So Modbus protocol address is equal to the Holding Register Offset minus one.

If you want to read more see modbus spec page 7 - "4.4 MODBUS Addressing model"
https://www.modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf

[image]