aldas/modbus-tcp-client #87: Error "stream_socket_client(): unable to connect" -> exception preferred

gutschein, 2021-05-18:
Many days my scripts (request frequency 15sec) run smooth with your library. I catched all usual exceptions as the modbus devices sometimes don't answer: Everything as expected with a few somereadTimeout or connectTimeout over the day. Very nice! :)

Today I had an new error which wasn't an exceptions and hence I couldn't catch. I would appreciate to get an exceptions instead:

`.../modbus-tcp-client/vendor/aldas/modbus-tcp-client/src/Network/InternetDomainStreamCreator.php | Line 39 | stream_socket_client(): unable to connect to 192.168.xxx.yyy:502 (Connection timed out)`

aldas, 2021-05-21:
mmm, you reporting that if you change for example ip to 8.8.8.8 in this example https://github.com/aldas/modbus-tcp-client/blob/master/examples/fc3.php it would not end up in exception block?

gutschein, 2021-05-21:
indeed: In this case I get both: An Exception I can catch AND an error
```
Catched: Unable to create client socket to 8.8.8.8:502: Connection timed out
ERROR: stream_socket_client(): unable to connect to 8.8.8.8:502 (Connection timed out)
```

aldas, 2021-05-21:
check your php configuration related to errors(reporting)

gutschein, 2021-05-21:
I do catch Exception. I don't handle errors by now. As they are not expected... :)
```
try {
	$fc3 = ReadRegistersBuilder::newReadHoldingRegisters()
		->allFromArray($registers)
		->build();
} catch (Exception $e) {
	echo('EXCEPTION Registers! '.$e->getMessage());
	return;
}
do {
	try {
		$response = (new NonBlockingClient(['readTimeoutSec' => $to_read, 'connectTimeoutSec' => $to_connect]))->sendRequests($fc3_part);
		break;
	} catch (Exception $e) {
		echo('EXCEPTION Request!'.$e->getMessage());
		return;
	}
	usleep(15000*1000);
}
```

aldas, 2021-05-21:
As I showed. That part uses `@` [operator suppress errors](https://www.php.net/manual/en/language.operators.errorcontrol.php). If it does not work - you have probably non-default PHP configuration.  

if you are returning then exception is catched it is expected that your function/scripts stops working.

gutschein, 2021-05-21:
I thought actually that the library would handle all errors to generate exceptions, which can be catched by the library users. Hence I didn't implemented an error handler. I thought you would appreciate to know and to fix it in the library. 

Probably that was a missunderstanding on my side. If it is more usual that the library useres do so - shouldn't be a big deal. I can add it into my code.

aldas, 2021-05-21:
I do not know what configuration you are running - using plain 7.4 or 8.0 does not have problems you have. As by default this is not a problem I see no reason to mess with error handlers during calls because by default you have none.

gutschein, 2021-05-21:
integrated an error_handler. errors are catched now as well, e.g. with testcase IP = 8.8.8.8

Thank you. 

Issue solved and closed

closed 2021-05-21 by gutschein