aldas/modbus-tcp-client #88: decode modBus binary stream

skydiablo, 2021-05-19:
hi,
i'm not really handy with modBus but i have already received the binary stream data via LoRaWAN. how can i decode this stuff to a more readable value? is modBus already defined in all details so i do not have to define an data-template or what is to do?

thx,
volker.

aldas, 2021-05-21:
could you please describe your setup more closely. modbus is request/response protocol and term `stream` confuses me a little. You can decode modbus response packets but you still would have raw register data and this needs to be unpacked to separate address values - but this information how to do it is not contained in response. You would receive for example 4 registers of data but wther you should read 4 int16 values or 2 int32 or float there is up to you decide.

skydiablo, 2021-06-03:
this is my example payload:

    $payload = base64_decode('DAYnARUADBRjaVEACy0AAAALOwIA8ApaVAIKXlcCDHhUOVhnAv0XAAAA');

and this is the payload definition: https://www.elvaco.se/Image/GetDocument/en/286/cmi4110-users-manual-english.pdf#page=15

so can i use this library to decode the payload?

aldas, 2021-06-03:
This manual/document talks about M-Bus. M-Bus (Meter-Bus) and Modbus are different things. From that document it seems that you could set it to send even JSON. In that case it would be very easy to convert into php objects/arrays.

skydiablo, 2021-06-03:
ahhhh !!!! M-Bus -vs- ModBus !!! okay, i see... yes, this module supports already JSON, but there is just a little peace of information to use the M-Bus style. so i have to decode the M-Bus style and i have it solved it just now by my own. so big thx for help and clear the missunderstood between "M-Bus" and "ModBus".

closed 2021-06-03 by skydiablo