aldas/modbus-tcp-client #89: Exception  with "Endian::LITTLE_ENDIAN_LOW_WORD_FIRST"

gutschein, 2021-05-26:
Hi Martti,

`Endian::$defaultEndian = Endian::LITTLE_ENDIAN_LOW_WORD_FIRST;`

I get an exception "Undefined class constant 'LITTLE_ENDIAN_LOW_WORD_FIRST'" Do I have a typo or a missunderstanding about the 4th possible endian option? The other 3 work as expected.

aldas, 2021-05-27:
Endian is actually bitmask that consist of  byte order (LE or BE) and word order (low word first or "high word first")

See this:
https://github.com/aldas/modbus-tcp-client/blob/1fb85de630c45154c015f779c0785987d54c70e1/src/Utils/Endian.php#L44

If you want to set LE+LWF use:
```
Endian::$defaultEndian = Endian::LITTLE_ENDIAN | Endian::LOW_WORD_FIRST;
```

gutschein, 2021-05-28:
Usage wasn't clear for me in the help; i assumed that it wold work as for BIG_ENDIAN.

But with your hint: Works now as expetced! :) Thank you.