aldas/modbus-tcp-client #93: error code is different

anditsung, 2021-06-18:
on dev using mac
61 | Unable to create client socket to tcp://192.168.11.190:502: Connection refused

on prod using debian
111 | Unable to create client socket to tcp://192.168.11.190:502: Connection refused

using "aldas/modbus-tcp-client": "^2.3",

cant catch the correct problem on this

aldas, 2021-06-18:
Does it work when trying with `modpoll` for example? See: https://github.com/aldas/modbus-tcp-client/wiki/Debugging-guide#2-try-another-modbus-client or https://www.modbusdriver.com/modpoll.html

anditsung, 2021-06-18:
```
./modpoll -m tcp -a 1 -r 1 -c 13 192.168.11.190
modpoll 3.10 - FieldTalk(tm) Modbus(R) Master Simulator
Copyright (c) 2002-2021 proconX Pty Ltd
Visit https://www.modbusdriver.com for Modbus libraries and tools.

Protocol configuration: MODBUS/TCP, FC3
Slave configuration...: address = 1, start reference = 1, count = 13
Communication.........: 192.168.11.190, port 502, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave... (Ctrl-C to stop)
[1]: 0
[2]: 31036
[3]: 0
[4]: 113
[5]: 195
[6]: 216
[7]: 600
[8]: 604
[9]: 521
[10]: 1270
[11]: 312
[12]: 989
[13]: 963
-- Polling slave... (Ctrl-C to stop)
[1]: 0
[2]: 31036
[3]: 0
[4]: 113
[5]: 195
[6]: 216
[7]: 600
[8]: 604
[9]: 521
[10]: 1270
[11]: 312
[12]: 989
[13]: 963
-- Polling slave... (Ctrl-C to stop)```

dont have problem when accessing using modpoll.

the error code seems diff
on dev the error code
60 for connection time out
61 for connection refused

on prod the error code was
110 for connection time out
111 for connection refused

right now im using array to catch the error code

sometimes using the package i have connection refused error

aldas, 2021-06-18:
Error `errNo` depens on operating system you use - you can not rely on that much. Even error message could be operating system or PHP version specific. You should catch IOException instead - and assume thing just by that - ala you need to retry:

```php
} catch (IOException $e) {
// ...
}

anditsung, 2021-06-18:
i have try to test using IOException
still have the same error code

but i found this documentation regarding error code on socket client

https://aakinshin.net/posts/how-socket-error-codes-depend-on-runtime-and-operating-system/

closed 2021-06-18 by anditsung