aldas/modbus-tcp-client #100: Explain the the data that we get

newfaboc, 2021-11-29:
Hi aldas,
I'm a newbie for PLC. I have try modbus-tcp-lient and it sucess get the data. But i don't know how to read the data.
I want read the temperature of chiller. i have read the document of modbus. 

here is the part of the document (content ): 
- **Input register**   (Function code 0x04 supported)
```
**ADDRESS**        **DESCRIPTION**                         **VALUE**        **NOTE**
32004           Ambient air temperature Actual   value×10      For instance: 0x00C8=200=20.0°C
32005           Water inlet temperature Actual   value×10      For instance: 0x00C8=200=20.0°C  
```  

- **Holding register**   (Function code supported 0x03/0x06/0x10)
```
**ADDRESS**        **DESCRIPTION**                         **VALUE**                   **NOTE**
40003           Water inlet temperature in cooling mode     -50~250=-5~25°C          Same as 32004
40004           Water inlet temperature in heating mode      250~600=25~60°C         Same as32004
40005           Water tank temperature setting               250~500=25~50°C         Same as32004
```

I want Get the address 40003 and 40004. 

may you help me ? what should i do to get it ?

Thank you

[image]

aldas, 2021-11-29:
Address `32004` means that it should be requested with FC3 (Read Holding Registers) and from address `2004`
Address `40003` means that FC4 should be used and set the start address as `3`.

`examples/index.php` is using FC3 for request but you can change it to FC4 by
```php
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $unitId);
```
to
```php
$packet = new ReadInputRegistersRequest($startAddress, $quantity, $unitId);
```
NB: make sure to import that class also `use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;`

I assume that `40003` is `int8` type as that value may contain negative values. `40004` and `40005` are probably `uint8` only positive values.

This is modified version of `examples/fc4.php`

```php
<?php

use ModbusTcpClient\Network\BinaryStreamConnection;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersRequest;
use ModbusTcpClient\Packet\ModbusFunction\ReadInputRegistersResponse;
use ModbusTcpClient\Packet\ResponseFactory;

require __DIR__ . '/../vendor/autoload.php';

$connection = BinaryStreamConnection::getBuilder()
    ->setPort(502)
    ->setHost('127.0.0.1')
    ->build();

$startAddress = 3;
$quantity = 3;
$packet = new ReadInputRegistersRequest($startAddress, $quantity);
echo 'Packet to be sent (in hex): ' . $packet->toHex() . PHP_EOL;

try {
    $binaryData = $connection->connect()
        ->sendAndReceive($packet);
    echo 'Binary received (in hex):   ' . unpack('H*', $binaryData)[1] . PHP_EOL;

    /**
     * @var $response ReadInputRegistersResponse
     */
    $response = ResponseFactory::parseResponseOrThrow($binaryData);
    echo 'Parsed packet (in hex):     ' . $response->toHex() . PHP_EOL;
    echo 'Data parsed from packet (bytes):' . PHP_EOL;
    print_r($response->getData());

    // set internal index to match start address to simplify array access
    $responseWithStartAddress = $response->withStartAddress($startAddress);
    print_r($responseWithStartAddress[3]->getInt16()); // 40003
    print_r($responseWithStartAddress[4]->getUInt16()); // 40004
    print_r($responseWithStartAddress[5]->getUInt16()); // 40005

} catch (Exception $exception) {
    echo 'An exception occurred' . PHP_EOL;
    echo $exception->getMessage() . PHP_EOL;
    echo $exception->getTraceAsString() . PHP_EOL;
} finally {
    $connection->close();
}
```

---------

NB: your images seems to reference CRC this is implies that RTU is used but I do not understand how in this case you can get valid response with `examples/index.php`? Does it support TCP?

-------

Bits of register can be checked with `print_r($responseWithStartAddress[5]->isBitSet(15)); // is register 40005 bit 15 set?`

aldas, 2021-11-29:
Note if you need only 0-14 bits from register you can use
```php
print_r($responseWithStartAddress[4]->getUInt16() >> 1); // get 0-14 bits of register 40004 ( `>> 1` discards 15th bit)
```

newfaboc, 2021-11-29:
NB: your images seems to reference CRC this is implies that RTU is used but I do not understand how in this case you can get valid response with examples/index.php? Does it support TCP?
-> i think yes. Like the image of exampel fc3. I enter the ip address of plc and it is response.

newfaboc, 2021-11-29:
ok

newfaboc, 2021-11-29:
Have you an email ? i will send a document if you want to read it.
Thank you aldas.

aldas, 2021-11-29:
nope :) but try with FC3 and FC4 starting from address `0`.  It could be that documentation has some copy-paste error and that `40001` maybe needs to be `30001`. 

One thing that I forgot -  when documents talk about `40001` it would mean that address `1` should be `0` in this library. Modbus docs usually reference addresses starting from `1` but in packet level addresses start from `0`  so you need to subtract 1.

newfaboc, 2021-11-29:
i have try the modified the example/fc4.php

Packet to be sent (in hex): 2f5d00000006040400030003
Binary received (in hex): 2f5d00000009040406433031325631
Parsed packet (in hex): 2f5d00000009040406433031325631
Data parsed from packet (bytes):
Array ( [0] => 67 [1] => 48 [2] => 49 [3] => 50 [4] => 86 [5] => 49 )
responseWithStartAddress[3] -> 17200
responseWithStartAddress[4] -> 12594
responseWithStartAddress[4] -> 22065 

i dont know how to read this. because i see the chiller the temperature "water in -> 11.2 C" and "the water out about -> 14.4 C".

May you help me ? what should next i do ?

> modified version of `examples/fc4.php`

aldas, 2021-11-29:
dec 22065 -> hex 0x56 0x31-> binary 0101 0110 0011 0001  <-- does not seem to have 15th bit as 1

assuming that 15th bit shows if value is valid then you need to get 0-14 bits. Try this. 
```php
    print_r($responseWithStartAddress[3]->isBitSet(15)); // is register 40003 valid?
    print_r($responseWithStartAddress[3]->getUInt16() & 0x7FFF); // 40003 first 14 bits

    print_r($responseWithStartAddress[4]->getUInt16() & 0x7FFF); // 40004 first 14 bits
    print_r($responseWithStartAddress[5]->getUInt16() & 0x7FFF); // 40005 first 14 bits
```

newfaboc, 2021-11-29:
> ```
> print_r($responseWithStartAddress[3]->isBitSet(15)); // is register 40003 valid?
>     print_r($responseWithStartAddress[3]->getUInt16() & 0x7FFF); // 40003 first 14 bits [...]

Packet to be sent (in hex): 293e00000006040400030003
Binary received (in hex): 293e00000009040406433031325631
Parsed packet (in hex): 293e00000009040406433031325631
Data parsed from packet (bytes):
Array ( [0] => 67 [1] => 48 [2] => 49 [3] => 50 [4] => 86 [5] => 49 )
isvalid ->
responseWithStartAddress[3] -> 17200
responseWithStartAddress[4] -> 12594
responseWithStartAddress[5] -> 22065 

the value of "print_r($responseWithStartAddress[3]->isBitSet(15)); // is register 40003 valid? " --> empty

aldas, 2021-11-29:
I have no idea. Maybe it is endianess. By default we use big endian but maybe these values are in Little endian.

This is same thing with little endian
```php
// should be added after reuires statements
Endian::$defaultEndian = Endian::LITTLE_ENDIAN;
```

or 
```php
    print_r($responseWithStartAddress[3]->getUInt16(2) & 0x7FFF); // 40003 first 14 bits

    print_r($responseWithStartAddress[4]->getUInt16(2) & 0x7FFF); // 40004 first 14 bits
    print_r($responseWithStartAddress[5]->getUInt16(2) & 0x7FFF); // 40005 first 14 bits
```

newfaboc, 2021-11-29:
> ```
> print_r($responseWithStartAddress[3]->getUInt16(2) & 0x7FFF); // 40003 first 14 bits
> 
>     print_r($responseWithStartAddress[4]->getUInt16(2) & 0x7FFF); // 40004 first 14 bits
>     print_r($responseWithStartAddress[5]->getUInt16(2) & 0x7FFF); // 40005 first 14 bits
> ```

i have try this and the result is :

Little Endian :
responseWithStartAddress[3] -> 12355
responseWithStartAddress[4] -> 12849
responseWithStartAddress[5] -> 12630

what it means ? 
I dont know why i have send again after several time it doesnt change.

aldas, 2021-11-29:
no idea but try FC3 also with same addresses ($startAddress = 0; $quantity = 3). with and without little endian. 

```php
    print_r($responseWithStartAddress[0]->getUInt16(2) & 0x7FFF); // 40003 first 14 bits  / little endian
    print_r($responseWithStartAddress[1]->getUInt16(2) & 0x7FFF); // 40004 first 14 bits
    print_r($responseWithStartAddress[2]->getUInt16(2) & 0x7FFF); // 40005 first 14 bits

    print_r($responseWithStartAddress[0]->getUInt16(1) & 0x7FFF); // 40003 first 14 bits / big endian
    print_r($responseWithStartAddress[1]->getUInt16(1) & 0x7FFF); // 40004 first 14 bits
    print_r($responseWithStartAddress[2]->getUInt16(1) & 0x7FFF); // 40005 first 14 bits
```

newfaboc, 2021-11-29:
> ```
> print_r($responseWithStartAddress[0]->getUInt16(2) & 0x7FFF); // 40003 first 14 bits  / little endian
>     print_r($responseWithStartAddress[1]->getUInt16(2) & 0x7FFF); // 40004 first 14 bits [...]

the result is : 
little endian :
responseWithStartAddress[0] -> 256
responseWithStartAddress[1] -> 25600
responseWithStartAddress[2] -> 16717
Big Endian :
responseWithStartAddress[0] -> 1
responseWithStartAddress[1] -> 100
responseWithStartAddress[2] -> 19777

the value doesnt change after i send 1 minute.

newfaboc, 2021-11-30:
hi aldas,
after i change From "`print_r($responseWithStartAddress[3]->isBitSet(15)); // is register 40003 valid?`"
to 
`print_r($responseWithStartAddress[3]->isBitSet(14)); // is register 40003 valid?`

it return 1.
it means 14 bit ?

aldas, 2021-11-30:
You are correct. I got this part wrong. The method implementation can be seen here.

https://github.com/aldas/modbus-tcp-client/blob/c39a0e51519c5ef76f6ff0f2fb8e36fb34de2935/src/Utils/Types.php#L236-L244

aldas, 2021-12-27:
@newfaboc you got it working? can I close the issue?

newfaboc, 2021-12-30:
yes. it works @aldas .
Thank you very much.

closed 2021-12-31 by aldas