aldas/modbus-tcp-client #127: Question: Trying to retrieve simple data through modbus

XavegX367, 2022-12-07:
Hello, I just started working with Modbus and I am using a program called 'Advantech Adam/Apax' Here I am able to turn on a light and count the amount of times a button has been pressed. Currently I am trying to retrieve this count and controll the light in php. Do you have any idea what I am doing wrong?

I am working with the Laravel framework, here is the function:

`public function index()
    {
        $connection = BinaryStreamConnection::getBuilder()
            ->setPort(502)
            ->setHost('192.168.250.50')
            ->build();

        $startAddress = 00001;
        $quantity = 1;

        $packet = new ReadCoilsRequest($startAddress, $quantity);

        try {
            $binaryData = $connection->connect()
                ->sendAndReceive($packet);

            $response = ResponseFactory::parseResponseOrThrow($binaryData);

            dd($response);
        } catch (Exception $exception) {
            dd($exception->getMessage() . PHP_EOL);
        } finally {
            $connection->close();
        }

    }`

Here is what the modbus looks like in adam:

[image]

Location: 00001 with description DI 0 is the button
Location: 00017 with description DO 0 is the light

I am only trying to retrieve the button count with the code above. But it only returns an array with a length of 8:

[image]

I am also curious about how I will be able to write to the coil so the light will turn on / off.

Thanks in advance.

aldas, 2022-12-07:
Please log raw bytes as https://github.com/aldas/modbus-tcp-client/blob/master/examples/fc1.php does. It is easier to examplain when I see what was sent and received. 

Only 1 "coil" is true because you asked for 1 coil `$quantity = 1;`. There are 8 coils in that class because FC1 response can at minimum reposond with 1 byte (8 bits/coils) where coils 1-7 are `0` zeroed.

XavegX367, 2022-12-08:
So I followed the example and now location 00001 returns true,  while I need it to return the times a button is pressed. Like shown in the image below:

[image]

This one should be under location DI[0] / DI 0. On the previous post it is shown on location 00001

[image]

I am also trying to get the state of a light this should be on DO[0 / DO 0 and this one should be on location 00017.
But when I turn the light on and try to receive 00017 it shows false while in the program it returns 1 (true).

aldas, 2022-12-08:
Did you try 0-based addresses? And please add sent/received packets as hex. 

At the moment I understand that when you read coil `17` value you get `0`.  When using some other tool that coil `17` is shown as `1`?

Do you use address as `17` or `16`?

XavegX367, 2022-12-08:
This is what it returns:

[image]

With this code:

[image]

XavegX367, 2022-12-08:
> Did you try 0-based addresses? And please add sent/received packets as hex.
> 
> At the moment I understand that when you read coil `17` value you get `0`. When using some other tool that coil `17` is shown as `1`?
> 
> Do you use address as `17` or `16`?
> 
> Please see https://modbus.org/docs/Modbus_Application_Protocol_V1_1b3.pdf page 7 "4.4 MODBUS Addressing model"

Indeed, when I use the tool on my windows machine it says bit: 1 on 00017.

aldas, 2022-12-08:
`$startAddress = 1; ` means that you request second address to ninth address. As address `1` in modbus model is actually PDU address `0`. This library uses PDU/0-based addresses.

That response bytes explained:
 * `\xb9\x02` - transaction id
 * `\x00\x00` - protocol id
 * `\x00\x04` - number of bytes in the message (PDU = ProtocolDataUnit) to follow
 * `\x00` - unit id
 * `\x01` - function code
 * `\x01` - coils byte count
 * `\x1f` - coils data (1 bytes = 1 * 8 coils)  hex `1f` => binary `0001_1111`

aldas, 2022-12-08:
If you are investigating address `0017` why do you request 8 coils starting from `0001`? You probably should move start address or quantity to see from this library side what is in those addresses.

```php
        $startAddress = 16;  // is 0-based address for 17
        $quantity = 1;
```

XavegX367, 2022-12-08:
> I doubt you can read amount of clicks with coils. Coil is binary (0 or 1). You probably need to use Register functions for this purpose. FC3 maybe http://www.simplymodbus.ca/FC03.htm

Yes, that works. I am getting the response in the last array. Thanks a lot!

closed 2022-12-08 by XavegX367