aldas/modbus-tcp-client #128: packet length more bytes than expected error

ranjithphptech, 2023-01-25:
<pre>root@ubuntu:/home/rkphptech# php /var/www/html/modbus-tcp-client-master/examples/write_usb_serial.php
RTU Binary to sent (in hex):   0106000d0001d9c9
2023-01-25 10:12:55.876190: [debug] msg: &quot;Connected&quot;
2023-01-25 10:12:55.880604: [debug] msg: &quot;Data sent&quot;, data: array (
  1 =&gt; &apos;0106000d0001d9c9&apos;,
)
2023-01-25 10:12:55.917155: [debug] msg: &quot;Polling data&quot;
2023-01-25 10:12:55.917287: [debug] msg: &quot;Stream 37 @ index: 0 received data: &quot;, data: array (
  1 =&gt; &apos;0106000d0001d9c9&apos;,
)
An exception occurred
packet length more bytes than expected
#0 /var/www/html/modbus-tcp-client-master/examples/write_usb_serial.php(16): ModbusTcpClient\Utils\Packet::isCompleteLengthRTU()
#1 /var/www/html/modbus-tcp-client-master/src/Network/StreamHandler.php(78): {closure}()
#2 /var/www/html/modbus-tcp-client-master/src/Network/BinaryStreamConnection.php(60): ModbusTcpClient\Network\BinaryStreamConnection-&gt;receiveFrom()
#3 /var/www/html/modbus-tcp-client-master/src/Network/BinaryStreamConnection.php(83): ModbusTcpClient\Network\BinaryStreamConnection-&gt;receive()
#4 /var/www/html/modbus-tcp-client-master/examples/write_usb_serial.php(32): ModbusTcpClient\Network\BinaryStreamConnection-&gt;sendAndReceive()
#5 {main}
</pre>

aldas, 2023-01-25:
Seems that ` Packet::isCompleteLengthRTU` is not suitable for write* responses.

This should work for FC6 response:
```php
->setIsCompleteCallback(static function ($binaryData, $streamIndex): bool {
return strlen($binaryData) != 6;
})
```

FC6 RTU response has 4 bytes of data + 2 bytes for CRC
https://github.com/aldas/modbus-tcp-client/blob/48c46a8327667cbe25a59a33078eff2d6d482748/src/Packet/ModbusFunction/WriteSingleRegisterResponse.php#L17

aldas, 2023-01-25:
Hey, I have released new version  `3.1.1` that fixes `Packet::isCompleteLengthRTU` to understand write response length correctly

aldas, 2023-01-26:
@ranjithphptech could you recheck with latest version?

closed 2023-02-17 by aldas