aldas/modbus-tcp-client #143: Question about data types

monkhaze, 2023-06-12:
Sorry in advance if this a stupid question.

I'm trying to read an input register from my inverter. The documentation says that the data type that it returns is an Uint32.
However when I run:
```
$binaryData = $connection->connect()->sendAndReceive($packet);
$response = ResponseFactory::parseResponseOrThrow($binaryData);

foreach ($response as $word) {
    print_r(get_class_methods($word));
}
```
I basically see that I can only get:
    [0] => getInt16
    [1] => getUInt16

When I try either of those the data doesn't seem to make any sense. When I'm getting a register that returns UInt16 for example everything seems to work fine.

How would I be able to get the data to return as UInt32 in this context?

Environment:
* OS: Ubuntu 22.04.2 LTS
* PHP version: PHP 8.1.18 (cli)
* Library version: aldas/modbus-tcp-client ^2.3

aldas, 2023-06-12:
Hi, 

You are seeying `getInt16` and `getUInt16` because `ReadHoldingRegistersResponse` implement iterator this way

https://github.com/aldas/modbus-tcp-client/blob/09bd467b9821d394f4b29f5bf5b0984c005fbdb0/src/Packet/ModbusFunction/ReadHoldingRegistersResponse.php#L163-L166

meaning 
```php
foreach ($response as $word) {
    print_r(get_class_methods($word));
}
```

will return `Word` class instances which represent 16bit / 2 byte long data.

If you want to access 32bit / double word fields you can do

https://github.com/aldas/modbus-tcp-client/blob/09bd467b9821d394f4b29f5bf5b0984c005fbdb0/examples/fc3.php#L45-L48

on:
* line `45` you set the internal address to your requested $startAddress
* line `48` you get 32bit float value from that address. NB: `DoublWord` has method `getUInt32` which is suitable for you.

aldas, 2023-06-12:
https://github.com/aldas/modbus-tcp-client/blob/master/examples/index.php this example is handy for debugging different data types and has overview how different types can be accessed

monkhaze, 2023-06-12:
@aldas Thanks for the very detailed response! I got it to work now.

closed 2023-06-12 by monkhaze