aldas/modbus-tcp-client #159: Limitation of Unit ID

Winni66, 2024-02-22:
**Is your feature request related to a problem? Please describe.**
I'm quite new to Modbus topic. This library is used within one Smarthome solution called Edomi. I'm using a Webasto Wallbox which uses Unit ID 255 and unfortunately this library allowes only ID up to 247. It's not possible to change.

**Describe the solution you'd like**
Would it be possible to disable ID limitation? I'm not able to build my own library for that, because I don't have the skills.

aldas, 2024-02-22:
As a immediate workaround - you can change the unitID value after packet has been created.
```php
$packet = new ReadHoldingRegistersRequest($startAddress, $quantity, $unitID); 
$packet[6] = 255;
```

For example: unit ID is 7th byte for FC3 request.

https://github.com/aldas/modbus-tcp-client/blob/53657b3de8f90c9e6727c9aca841b3e355fbe554/src/Packet/ModbusFunction/ReadHoldingRegistersRequest.php#L21

Winni66, 2024-02-22:
Thanks for fast reply, but i fear it's not as easy as it seems for me:

 

1) I'm using this library for several modbus devices (wallbox, inverter, battery), therefore workaround with fixed ID will not work.

2) I tried to remove the limit here:
        private function validate(int $length, int $unitId, int|null $transactionId): void
    {
        if (!$length || !($length > 0 && $length <= Types::MAX_VALUE_UINT16)) {
            throw new InvalidArgumentException("length is not set or out of range (uint16): {$length}");
        }
        if (!($unitId >= 0 && $unitId <= 257)) {
            throw new InvalidArgumentException("unitId is out of range (0-257): {$unitId}");
but it doesn't has impact. Unfortunately I don't know what composer does on my installation, perhaps there is something like compiling? I don't know how to handle this properly.

 

best regards

 

P.S.: there is also a message:

Cannot use aldas/modbus-tcp-client's latest version 3.5.0 as it requires php ^8.0 which is not satisfied by your platform.

I'm still on 7.2.34

 

Gesendet: Donnerstag, 22. Februar 2024 um 23:22 Uhr

 

https://github.com/aldas/modbus-tcp-client/blob/53657b3de8f90c9e6727c9aca841b3e355fbe554/src/Packet/ModbusFunction/ReadHoldingRegistersRequest.php#L21

PR #160 (2024-03-10): Allow unit ID to be in range of 0-255.
Older MODBUS specification limited unit ID (slave id) in range of 0-247).  See "Modicon Modbus Protocol Reference Guide PI–MBUS–300 Rev. J" page 19. https://modbus.org/docs/PI_MBUS_300.pdf

In newer spec that limitation is removed.

Fixes #159
closed 2024-03-10 by aldas