aldas/modbus-tcp-client #164: Problema na conexão F04

denissonwenergy, 2024-11-01:
I saw your project on github: https://github.com/aldas/modbus-tcp-client.

I have a park with the following configuration: Teltonikas Gateways and energy meters connected via RS485.

I tried to use FC4 to connect to the IP+Port of the remote gateway (teltonika) and I get the connection refused error.

Unable to create client socket to tcp://177.xx.xxx.xxx:4000: Connection refused #0 /var/www/duodigit/energy/vendor/aldas/modbus-tcp-client/src/Network/BinaryStreamConnectionBuilder.php(33): ModbusTcpClient\Network\InternetDomainStreamCreator->createStream() #1 /var/www/duodigit/energy/ vendor/aldas/modbus-tcp-client/src/Network/BinaryStreamConnection.php(42): ModbusTcpClient\Network\BinaryStreamConnectionBuilder->ModbusTcpClient\Network\{closure}() #2 /var/www/duodigit/energy/overip.php(137): ModbusTcpClient\Network\BinaryStreamConnection->connect() #3 /var/www/duodigit/energy/overip.php(40): getRegister() #4 /var/www/duodigit/energy/vendor/evenement/evenement/src/EventEmitterTrait.php(143): {closure}() #5 /var/www/duodigit/energy/vendor/react/socket/src/Socket Server.php(76): Evenement\EventEmitter->emit() #6 /var/www/duodigit/energy/vendor/evenement/evenement/src/EventEmitterTrait.php(143): React\Socket\SocketServer->React\Socket\{closure}() #7 /var/www/duodigit/energy/vendor/react/socket/src/TcpServer.php(258): Evenement\EventEmitter->emit() #8 /var/www/duodigit/energy/vendor/react/socket/src/TcpServer.php(239): React\Socket\TcpServer->handleConnection() #9 /var/www/duodigit/energy/vendor/re act/event-loop/src/StreamSelectLoop.php(246): React\Socket\TcpServer->React\Socket\{closure}() #10 /var/www/duodigit/energy/vendor/react/event-loop/src/StreamSelectLoop.php(213): React\EventLoop\StreamSelectLoop->waitForStreamActivity() #11 /var/www/duodigit/energy/overip.php(119): React\EventLoop\StreamSelectLoop->run()
#12 {main}

I would be very grateful for your help.

* Environment:
 - OS: [e.g. Ubuntu20.10/...]
 - PHP version > 8

aldas, 2024-11-02:
`Connection refused` implies that Server is not accepting connections. This is probably due to the firewall. 

Teltonika guide says
> To test the configuration, we could try to change some values in the Modbus PLC simulator and then try to read them via Modbus Master application. **You might need to disable Firewall in both PC1 and PC2, or modify Firewall rules to allow both applications through the Firewall.**

https://wiki.teltonika-networks.com/view/Data_transfer_from_Modbus_TCP_to_Modbus_RTU_Gateway#Testing_the_configuration

Firewall configuration guide is here https://wiki.teltonika-networks.com/view/Firewall_traffic_rules

also there is that section
> In IP filter options on the bottom of the page, add IP addresses that will be allowed to reach your Modbus gateway (input 0.0.0.0/0 to allow any IP address).

so you might need to allow your IP there

denissonwenergy, 2024-11-03:
Good morning, 

I am using a feature on the Teltonika TR145 called "Over IP", that is,
my application acts as a server, and the gateway connects to it and
after that, I need to request the data (which I imagine is via Modbus). 

Have you had any experience with this type of setup? 

Thank you for your time and support.

--- 

  

Links:
------
[1]
https://github.com/aldas/modbus-tcp-client/issues/164#issuecomment-2453046800
[2]
https://github.com/

aldas, 2024-11-03:
I have one Teltonika RUT956 at my desk at work and have tried in locally Modbus (mostly through their modbus->mqtt thingie)

denissonwenergy, 2024-11-03:
Good morning, 

I'm trying to use the feature: Services -> OverIP, leaving Teltnika as
*client*, but I haven't been successful in using the feature yet. 

My application, socket server, receives the connection from the device,
so I have the IP and Port. I try to make the connection from the data I
receive, but I'm not successful, even though the firewall rules are
enabled. 

I've already tested the feature: Data to Server, from Teltnika, and I'm
successful in that, but the data consumption is very high, ending my
mobile data allowance from the M2M chip. 

  

Links:
------
[1]
https://github.com/aldas/modbus-tcp-client/issues/164#issuecomment-2453446651
[2]
https://github.com/

aldas, 2024-11-03:
Could you help me little bit to understand your setup.

You have
1. energy meters that are connected to Teltonika routers over RS485
2. Teltonika routers at the "edge"
3. Server at the "internet"

You have tried https://wiki.teltonika-networks.com/view/RUT200_Data_to_Server but this is not succient

You are now trying to use this library - by connecting periodically from 3. (from Server) to 2. (Teltonika) and your request end up in `Connection refused`?

Do you use https://wiki.teltonika-networks.com/view/RUT200_Modbus to expose your RS485 device as TCP port?

denissonwenergy, 2024-11-03:
Good morning, 

Exactly. 

I have energy meters connected to Teltonika routers; 

I am using the option (in Teltonika): Services -> Over IP (via TCP), in
client mode, to connect to my remote server because another software
vendor was already using the same feature. 

https://wiki.teltonika-networks.com/view/Serial_OverIP_communication_between_Server_and_Client

It seems to me that, in the current configuration, the Teltonika
routers, which have the current configuration, connect to my server
application (and I can check the IPs that are arriving). My question is:
How do I connect to the Teltonika routers and request the data (from the
registers) after receiving this connection? Is it via Modbus or is there
another way? 

I am attaching a printout of my configuration.

--- 

  

Links:
------
[1]
https://github.com/aldas/modbus-tcp-client/issues/164#issuecomment-2453452639
[2]
https://github.com/

closed 2025-11-20 by aldas