aldas/modbus-tcp-client #168: The returned data is misaligned with the actual data

IamComing3, 2024-12-17:
Hi

When I use the following code to get the data：

```
$address = 'tcp://192.168.2.228:502';
$unitID = 1;

$register = ReadRegistersBuilder::newReadInputRegisters($address, $unitID); 
$register ->float(1, 'a30001a');
$register ->float(3, 'a30003a');
$register ->float(5, 'a30005a');
$register ->int16(7, 'a30007a');
$register ->int16(8, 'a30008a');

$req = $register ->build();

$responseContainer = (new NonBlockingClient(['readTimeoutSec' => 0.5]))->sendRequests($req);

return $responseContainer->getData();
```

The data obtained are as follows：

```
[
  "a30001a" => 6.9375
  "a30003a" => 8.489914894104
  "a30005a" => 1.550985166284E-40
  "a30007a" => 0
  "a30008a" => 101
]
```

But the correct data should be：

```
[
  "a30001a" => ...
  "a30003a" => 6.9375
  "a30005a" => 8.489914894104
  "a30007a" => 1.550985166284E-40 
  "a30008a" => ...
]
```

I can't understand why

aldas, 2024-12-17:
Probably related to endian. Test these registers with https://github.com/aldas/modbus-tcp-client/blob/master/examples/index.php

aldas, 2024-12-17:
ok, never mind the previous comment.

Why do you think that `a30007a1` should be `1.550985166284E-40 ` which is actually returned in `a30005a`?

You are requesting `$register ->int16(7, 'a30007a');` as `int16` value.  But say that you want it to be float32 value?

IamComing3, 2024-12-17:
> ok, never mind the previous comment.
> 
> Why do you think that `a30007a1` should be `1.550985166284E-40 ` which is actually returned in `a30005a`?
> 
> You are requesting `$register ->int16(7, 'a30007a');` as `int16` value. But say that you want it to be `float32` value?

Thanks for the reply
I may have misunderstood, but for sure “a30003a” => 6.9375, “a30005a” => 8.489914894104, this result is obtained by ModScan

IamComing3, 2024-12-17:
> ok, never mind the previous comment.
> 
> Why do you think that `a30007a1` should be `1.550985166284E-40 ` which is actually returned in `a30005a`?
> 
> You are requesting `$register ->int16(7, 'a30007a');` as `int16` value. But say that you want it to be `float32` value? To me receiving ` "a30007a" => 0, "a30008a" => 101` seems OK because these 2 values are requested as int16 values.
> 
> ```
> $register ->int16(7, 'a30007a');
> $register ->int16(8, 'a30008a');
> ```

And I tried to set a different Endian for flaot and int16 respectively, but the results are wrong, and the only way to get the closest results is to set it to BIG_ENDIAN_LOW_WORD_FIRST

closed 2024-12-17 by IamComing3

aldas, 2024-12-17:
> this result is obtained by ModScan

just for information. The addresses this library uses are PDU/ protocol addresses that start from 0. Often Modbus aadresses are written in documentation in 1-based form.  So when your documentation is 1-based and your data is at register `10`  you need to write that register address as `9`   

For for "Modbus: When 40001 Really Means 1, or 0 Really Means 1" block in https://www.csimn.com/CSI_pages/Modbus101.html  it describes 0based and 1-based addresses.

IamComing3, 2024-12-17:
> > this result is obtained by ModScan
> 
> just for information. The addresses this library uses are PDU/ protocol addresses that start from 0. Often Modbus aadresses are written in documentation in 1-based form. So when your documentation is 1-based and your data is at register `10` you need to write that register address as `9` for this library
> 
> See "Modbus: When 40001 Really Means 1, or 0 Really Means 1" block in https://www.csimn.com/CSI_pages/Modbus101.html it describes 0based and 1-based addresses.

Yeah, it's settled. Anyway, you're my hero bro