aldas/modbus-tcp-client #169: Why do you get 16 results when the quantity is set to 10?

IamComing3, 2024-12-23:
* Environment:
 - PHP version 8.1
 - Library version 3.5.1

hello there.

I use the following code to read data

```
    Endian::$defaultEndian = Endian::BIG_ENDIAN;

    $connection = BinaryStreamConnection::getBuilder()
        ->setPort(502)
        ->setHost('192.168.0.99')
        ->setConnectTimeoutSec(1.5) 
        ->setReadTimeoutSec(0.3)
        ->build();

    $startAddress = 0;
    $quantity = 10;
    $unitID = 1;
    $packet = new ReadInputDiscretesRequest($startAddress, $quantity, $unitID); 

    $binaryData = $connection->connect()->sendAndReceive($packet);

    $response = ResponseFactory::parseResponseOrThrow($binaryData);

    return $response->getCoils();
```

The result obtained is this

```
array:16 [▼ // routes\web.php:101
  0 => true
  1 => true
  2 => true
  3 => true
  4 => true
  5 => true
  6 => false
  7 => true
  8 => false
  9 => true
  10 => false
  11 => false
  12 => false
  13 => false
  14 => false
  15 => false
]
```

I understood the quantity to be the number of addresses that need to be read, did I misunderstand?

IamComing3, 2024-12-23:
Thank you very much, your replies are always so fast, I'll go through the documentation first, if I can't fix it I'll go back and continue to reply, otherwise I'll close the issue, thanks again!

IamComing3, 2024-12-23:
Indeed, the return result is a multiple of 8. 

Assuming the starting address is 1, can I treat the first element of the return result as the result of 30001, the second element as the result of 30002, and so on,

because by comparing the data I found that the order of the results is the same as the order of the results I read out through modScan

aldas, 2024-12-23:
yes, this library uses protocol level address numbering scheme where first address is `0`.  This `0` is same as addresses given `20001` format.  

I do not know about your `30001` as usually first number is marks function code used. so FC3 is Read Holding Registers.  but ofter FC2 and FC3 aadresses map to same memory and you can query same data from FC2/FC3 but need to know how Coil address maps to Register address.

closed 2024-12-23 by IamComing3