aldas/modbus-tcp-client #171: How to get long integer data？

IamComing3, 2024-12-25:
Hi there.

I want to finally get the data of long integer type according to the following following code

```
$req = ReadRegistersBuilder::newReadInputRegisters($address, $unitID)
        ->int16(30033, 'a33')
        ->build();
```

But the final result did not meet expectations, so I rewrote it based on the method under this issue [143#issuecomment-1587668718](https://github.com/aldas/modbus-tcp-client/issues/143#issuecomment-1587668718)

```
$req = ReadRegistersBuilder::newReadInputRegisters($address, $unitID)
        ->int16(30033, 'a33', function ($value,  $addr,  $response) {
            $response_with_start = $response->withStartAddress(30033);
            $response_with_start[30033]->getBytes();
            return $response_with_start->getDoubleWordAt(30033)->getInt32();
        })
        ->build();
```

After rewriting, an error was captured: `Address out of bounds`

Then I tried using the `->int32(30033, 'a33')` and `->int64(30033, 'a33')`, but the result was incorrect

May I ask if there are any other methods available

IamComing3, 2024-12-25:
After testing, it is currently possible to read the data correctly using this method

```
$req = ReadRegistersBuilder::newReadInputRegisters($address, $unitID)
        ->int32(30001, 'a30001', fn ($value, $address, $response) => $response[30001]->getInt16())
        ->int32(30003, 'a30003', fn ($value, $address, $response) => $response[30003]->getInt16())
        ->build();
```

Of course, if you have any better suggestions, I hope you can reply to me. Thanks❤️

closed 2024-12-25 by IamComing3

aldas, 2024-12-25:
```php
->int32(30001, 'a30001', fn ($value, $address, $response) => $response[30001]->getInt16())
```

means that you are requesting 32bits of data (32 = 2 registers) and then getting only first 16 bits (1 register) out of it.   I think you have discovered the wonderful world of endianess and word order.

I recommend using https://github.com/aldas/modbus-tcp-client/blob/master/examples/index.php to see which endian you need to use to get multiregister values correct. Set the startaddress and quantity and try different endian options.

This depends on your device. This library default endian is `BIG_ENDIAN_LOW_WORD_FIRST` but you might want to to try just `BIG_ENDIAN`

```php
$req = ReadRegistersBuilder::newReadInputRegisters($address, $unitID)
    ->int32(30001, 'a30001', null, null, Endian::BIG_ENDIAN)
    ->build();
```

I have documented endianess a little bit here
https://github.com/aldas/modbus-tcp-client/blob/a11a0f98941cda0ec9a4f6ee4cef6721c1359863/src/Utils/Endian.php#L15-L25

Maybe read this https://www.digi.com/resources/documentation/digidocs/90001537/references/r_modbus_floating_points.htm